openscad/openscad #5990: multi core processing for CSG components relating to rendering objects.

FOSSBOSS, 2025-06-22:
Problem? no problem, its just so flipping slow. So slow, for every N+ object render time increases by N+.

**Describe the solution you'd like**
threading, using std::threads a feature of the standard library since C11 (2011)
The component which has the most to gain from threading is rendering code related to CSG.
specifically union, difference and intersection function child nodes.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Started reading the source code. might take me ages to get up to speed though. started looking at thread safe standard implementations. 

**Additional context**
models with increased complexity, or even just more parts take more time to render. I feel like everyone has at least 4 cores at this point, if not substantially more. 

concerns:
wont this use more memory? yep, sure would. what do?
limit thread count to std::thread::hardware_concurrency()
or less. If someone has less memory, or cores, use single core operation

 as a fallback. 

Benefits:
Cut rendering time in half for complex models, ... or maybe more. 

Legit, took an hour to render this. I have 10 cores and 64GB ram, it would be nice if I could utilize more of that. 
[image]

t-paul, 2025-06-22:
Dev Snapshot with Manifold enabled in Preferences can use multiple cores.

t-paul, 2025-06-22:
https://fosstodon.org/@OpenSCAD/113256867413539398

closed 2025-06-22 by t-paul