openscad/openscad #5991: add manifold rendering to initial object load

FOSSBOSS, 2025-06-22:
**Is your feature request related to a problem? Please describe.**
manifold rendering is much faster, but it seems to only be in effect on objects already loaded in openscad.
add this feature to objects on initial load.

**Describe the solution you'd like**
when opening an already generated object, use manifold rendering when manifold rendering is selected.

**Describe alternatives you've considered**
I do have a work around.
1. load a simple file.scad.
2. generate the complex object code
3. reload.
The reload will use the fast rendering manifold option. 
where if you generate a complex object, then try to load it directly, it will take absolutely forever to render.

this works up to 200K objects, before some other function breaks the work around
WARNING: Normalized tree is growing past 200000 elements. Aborting normalization.

WARNING: CSG normalization resulted in an empty tree

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

kintel, 2025-06-22:
Could you provide a concrete example? It sounds like you might be hitting a slow path of preview rendering. Manifold is either enabled or not; there is no concept of "already loaded" related to Manifold.

FOSSBOSS, 2025-06-22:
Sure. When I generate a scad file, using some code that converts pixels to height mapped cubes, 
I get a scad file with several thousand cube objects in a scad file. If I open the program output.scad, 
openscad will take forever to open the file, or crash x11.  But, if I open the output file, just as a blank file, that will open, then run the program to generate the scad model, and refresh, then the manifold rendering will work. 

Heres a script to generate scad from images:
https://github.com/FOSSBOSS/scads/blob/main/Scripts/img2scad.py

Here is an output file.
https://github.com/FOSSBOSS/scads/blob/main/Scripts/output1.scad

when openscad output1.scad
the file will take a long time to load. in excess of 2 minutes to load the gui, and possibly more to then render the image / model.
looking at cpu utilization: at no point will more than one core be utilized. so I think it is not using the manifold method on initial load / render. 

I can see where manifold rendering does use multiple cores, while that process is going on, but not when the file is already loaded.  

reloading the data if openscad is open already happens quickly. 

Dev build version 2025.6.15
maybe you would like a video? or some other proof? I can make this. but it would be a long boring video.

t-paul, 2025-06-22:
I believe it's supposed to *parse* the scad file(s) initially to the AST, mainly for finding variables to populate the customizer. Maybe that path somehow now triggers a full preview or render instead? Plus that may happen before the settings are initialized so it indeed could cause CGAL to still be used as that's the default.

kintel, 2025-06-22:
Workaround: Preferences->Editor->Color syntax highlighting: Off

pca006132, 2025-06-23:
It seems that the QScintilla is doing something crazy per `setStyling` call.

The following patch fixes the issue by doing a batch set styling. I don't have much time to clean up the other code (dead code now), so I will just drop it here and see anyone can pick this up. At least the GUI can now open in a few seconds after entering this lexer code, better than previous (I did not wait for it to complete, it feels too slow)

```diff
diff --git i/src/gui/ScadLexer.cc w/src/gui/ScadLexer.cc
index f6bff2fb2..d1238338e 100644
--- i/src/gui/ScadLexer.cc
+++ w/src/gui/ScadLexer.cc
@@ -205,11 +205,17 @@ void Lex::lex_results(const std::string& input, int start, LexInterface *const o
   int isstyle = obj->getStyleAt(start - 1);
   if (isstyle == ecomment) results.state = rules_.state("COMMENT");
 
+  char* styles = new char[input.size()];
+
   lexertl::lookup(sm, results);
   while (results.id != eEOF) {
-    obj->highlighting(start, input, results);
+    size_t s = std::distance(input.begin(), results.first);
+    size_t len = std::distance(results.first, results.second);
+    memset(styles + s, results.id, len);
     lexertl::lookup(sm, results);
   }
+  obj->highlightingMultiple(start, input.size(), styles);
+  delete [] styles;
 }
 
 /***************************************************************/
@@ -227,9 +233,9 @@ ScadLexer2::~ScadLexer2()
 
 void ScadLexer2::styleText(int start, int end)
 {
-#if DEBUG_LEXERTL
+// #if DEBUG_LEXERTL
   std::cout << "start: " << start << std::endl;
-#endif
+// #endif
   if (!editor()) return;
 
   char *data = new char[end - start + 1];
@@ -319,9 +325,14 @@ int ScadLexer2::getStyleAt(int pos)
   return sstyle;
 }
 
+void ScadLexer2::highlightingMultiple(int start, int length, char* styles) {
+  editor()->SendScintilla(QsciScintilla::SCI_STARTSTYLING, start);
+  editor()->SendScintilla(QsciScintilla::SCI_SETSTYLINGEX, length, styles);
+}
+
 void ScadLexer2::highlighting(int start, const std::string& input, lexertl::smatch results)
 {
-  std::string token = results.str();
+  // std::string token = results.str();
   int style = results.id;
 
 #if DEBUG_LEXERTL
@@ -330,7 +341,7 @@ void ScadLexer2::highlighting(int start, const std::string& input, lexertl::smat
 #endif
 
   startStyling(start + std::distance(input.begin(), results.first));
-  setStyling(token.length(), style);
+  setStyling(std::distance(results.first, results.second), style);
 }
 
 QColor ScadLexer2::defaultColor(int style) const
diff --git i/src/gui/ScadLexer.h w/src/gui/ScadLexer.h
index ac6191d6a..c80d58f93 100644
--- i/src/gui/ScadLexer.h
+++ w/src/gui/ScadLexer.h
@@ -51,6 +51,7 @@ class LexInterface
 {
 public:
   virtual void highlighting(int start, const std::string& input, lexertl::smatch results) = 0;
+  virtual void highlightingMultiple(int start, int length, char* styles) = 0;
   virtual int getStyleAt(int position) = 0;
 };
 
@@ -119,6 +120,7 @@ public:
   QColor defaultColor(int style) const override;
 
   void highlighting(int start, const std::string& input, lexertl::smatch results) override;
+  void highlightingMultiple(int start, int length, char* styles) override;
   QString description(int style) const override;
   QStringList autoCompletionWordSeparators() const override;
```

PR #6025 (2025-07-20): faster syntax highlighting
Closes #5991.

This performs batch highlighting by using the `SCI_SETSTYLINGEX` event. Previously, we were using one event per token for highlighting, which triggers rendering after every single token (apparently, from gdb stepping) and was very slow for large files. With the batch API, this is much faster and can open the 33MB file in less than 10s (most of the time is probably not in highlighting).
closed 2025-07-20 by t-paul