openscad/openscad #5994: Windows nightly build: export 3mf with color does not do so.

gtoal, 2025-06-23:
**Describe the bug**
I'm using the nightly build for Windows 64 bit. I have tried both back ends.  Export as 3mf with "Use colors from model and color scheme" selected and "Export colors as: color" selected exports the objects with only one colour encoded.  I have unzipped the .3mf file and checked the output.  Only one colorgroup and one colour is present. The object is built with two colours.

**Code reproducing the issue**
```
echo(version=version());

font = "Liberation Sans";
//["Liberation Sans", "Liberation Sans:style=Bold", "Liberation Sans:style=Italic", "Liberation Mono", "Liberation Serif"]

// all measurements are in mm
layerdepth = 0.4;
width = 25;
height = 10;
depth = 4*layerdepth;

module hello() {
  translate([0,0,-3*layerdepth]) linear_extrude(3*layerdepth) text("HELLOO", size=5);
}

difference() {
  union() {
    translate([1.5,3,0]) union() {

      difference() {
        color("black") translate([-1.5,-3,0]) cube([width,height,depth]);
        translate([0,0,depth]) hello();
      }

      color("white") translate([0,0,depth]) hello();
    }

    translate([1.5,height+3+1.2,0]) union() {

      difference() {
        union() {
          color("white") translate([-1.5,-3,0]) cube([width,height,depth-layerdepth]);
          color("black") translate([-1.5,-3,depth-layerdepth]) cube([width,height,layerdepth  ]);
        }
        translate([0,0,depth]) hello();
      }

      color("white") translate([0,0,depth]) hello();
    }
  }

  color("white") translate([width,0,0]) cube([width, height*2+1.2,depth+0.1]);
}
```

**Environment and Version info (please complete the following information):**
```
 - OS: Windows 10
 - System:
Processor	Intel(R) Core(TM) i7-6820HQ CPU @ 2.70GHz   2.70 GHz
Installed RAM	32.0 GB (31.9 GB usable)
Graphics Card	NVIDIA Quadro M2200 (4 GB), Intel(R) HD Graphics 530 (128 MB)
System Type	64-bit operating system, x64-based processor

 - OpenSCAD Version 2025.06.21 (git d4994a14b) from github today

**Library & Graphics card information**
OpenSCAD Version: 2025.06.21 (git d4994a14b)
System information: Microsoft Windows 10 (10.0.19045) x86_64 8 CPUs 31.85 GB RAM
User Agent: OpenSCAD/2025.06.21 (git d4994a14b) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
[... 287 lines cut ...]
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

```

Output starts with:
<?xml version="1.0" encoding="utf-8"?>
<model xmlns="http://schemas.microsoft.com/3dmanufacturing/core/2015/02" unit="millimeter" xml:lang="en-US" xmlns:m="http://schemas.microsoft.com/3dmanufacturing/material/2015/02" xmlns:p="
http://schemas.microsoft.com/3dmanufacturing/production/2015/06" xmlns:b="http://schemas.microsoft.com/3dmanufacturing/beamlattice/2017/02" xmlns:s="http://schemas.microsoft.com/3dmanufactu
ring/slice/2015/07">
        <metadata name="Title">Surface text experiments</metadata>
        <metadata name="Application">OpenSCAD (https://www.openscad.org/)</metadata>
        <metadata name="CreationDate">2025-06-23T03:54:02Z</metadata>
        <resources>
                <m:colorgroup id="2">
                        <m:color color="#DDDDFFFF" />
                </m:colorgroup>
                <object id="1" name="OpenSCAD Model" type="model" p:UUID="7f1d3856-fe72-4a97-9576-1f16b093df60" pid="2" pindex="0">
                        <mesh>
                                <vertices>
                                        <vertex x="0" y="0" z="0" />
...
(no other objects or colorgroups. just vertices and triangles. truncated for brevity.)

caufang, 2025-08-02:
@gtoal I took your example code and in OpenSCAD version 2025.07.11 `3MF` export is properly defining multiple color sections:

```xml
<?xml version="1.0" encoding="utf-8"?>
<model xmlns="http://schemas.microsoft.com/3dmanufacturing/core/2015/02" unit="millimeter" xml:lang="en-US" xmlns:m="http://schemas.microsoft.com/3dmanufacturing/material/2015/02" xmlns:p="http://schemas.microsoft.com/3dmanufacturing/production/2015/06" xmlns:b="http://schemas.microsoft.com/3dmanufacturing/beamlattice/2017/02" xmlns:s="http://schemas.microsoft.com/3dmanufacturing/slice/2015/07">
	<resources>
		<m:colorgroup id="2">
			<m:color color="#000000FF" />
			<m:color color="#FFFFFFFF" />
			<m:color color="#8ABEB7FF" />
		</m:colorgroup>
		<object id="1" name="OpenSCAD Model" type="model" p:UUID="9504e55b-7b86-47be-b475-3d9d501e16fd" pid="2" pindex="2">
			<mesh>
				<vertices>
					<vertex x="0" y="0" z="0" />
					<vertex x="0" y="0" z="1.600000" />
```

gtoal, 2025-08-03:
> [@gtoal](https://github.com/gtoal) I took your example code and in OpenSCAD version 2025.07.11 `3MF` export is properly defining multiple color sections:

Thanks for the notification! I'll try the latest nightly in that case (and close the thread as soon as it's confirmed).

kintel, 2025-08-14:
Works as expected on macOS from 2025.08.13.
Could be Windows-specific or related to lib3mf v1, so waiting for confirmation from the original reporter.

dgatwood, 2025-08-14:
For me, it turned out that the color data was actually there, and I missed it.  The problem was actually that the slicer didn't understand it.

closed 2025-10-03 by kintel