openscad/openscad #5995: Can't open library 'thread_profile.scad'

KungPhoo, 2025-06-23:
I can't open the library 'thread_profile.scad'.

I updated to the nightly 2025.06.21 (git d4994a14b), because of this problem:
https://github.com/openscad/openscad/issues/5443

the file is in the installation directory "C:\Program Files\OpenSCAD\libraries\thread_profile.scad".
However, I can reproduce the bug by creating a new file and typing:
```
use <thread_profile.scad>
```

I also tried <libraries/thread_profile.scad>, but that didn't work, either.

KungPhoo, 2025-06-23:
My bad - had two installatinos and only one had the libraries.

kintel, 2025-06-23:
I believe there's a shared folder where you can put libraries useable by multiple installations.