openscad/openscad #5999: Python venv created with AppImage can't be activated and used from CLI

RedKrieg, 2025-06-26:
**Describe the bug**
When a venv created using the "Create Virtual Environment" menu option (invoking [pythonCreateVenv](https://github.com/openscad/openscad/blob/25725172edaf7a10006590fe10e15757eb311516/src/python/pymod.cc#L79C5-L79C21)), the symlink for `bin/openscad-python` is set to `/proc/self/exe` and all other python shortcuts are linked to that.  When trying to install a Python library, it's common to use the command line to activate the virtualenv (`source bin/activate`) and then run executable scripts in the associated `bin` directory like `pip`, `python`, etc.  Unfortunately this does not behave as one might expect and instead runs the associated script (e.g. `bin/pip`) with the user's shell.  Since `bin/pip` in particular starts with something resembling valid shell script (some comments followed by `import re`) and `import` is a frequently installed window capture tool (from imagemagick), the result is the user being presented with a window capture cursor they may not have seen before, locking out any other input until they click, then a number of error messages that aren't relevant.

**To Reproduce**
Steps to reproduce the behavior:
1. Use OpenSCAD AppImage (latest nightly)
2. Install imagemagick to have `import` in your path
3. Create Virtual Environment in the UI, selecting a new empty directory.
4. In a bash terminal window from an X session, `cd` to the new virtual environment
5. Run `source bin/activate`
6. Run `pip freeze`
7. Your cursor will change and you must click twice before hitting a shell error.

**Expected behavior**
I would expect these commands to, if not succeed, then print a warning about being run in a specific way.  I've been working around this by invoking `~/Apps/OpenSCAD-2025.06.23.ai25983-x86_64_db127d35160c079a9fddbdbb1ac499cd.AppImage --python-module pip install [example]`.

**Code reproducing the issue**
```#!bash
source bin/activate
pip freeze
```

**Screenshots**
N/A

**Environment and Version info (please complete the following information):**
 - OS: Pop!_OS 22.04
 - System: Intel PC 64-bit
 - OpenSCAD-2025.06.23.ai25983-x86_64_db127d35160c079a9fddbdbb1ac499cd.AppImage

**Library & Graphics card information**
OpenSCAD Version: 2025.06.23.ai25983 (git 25725172e)
System information: Linux 6.12.10-76061203-generic #202412060638~1748542656~22.04~663e4dc SMP PREEMPT_DYNAMIC Thu M x86_64 Pop!_OS 22.04 LTS 24 CPUs 62.70 GB RAM
User Agent: OpenSCAD/2025.06.23.ai25983 (git 25725172e) (Linux x86_64; Pop!_OS 22.04 LTS)
Compiler: GCC "11.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>
Python Version: Python 3.10.12
OpenCSG version: OpenCSG 1.8.1
Clipper2 version: 1.5.2
Manifold version: 3.1.1
Qt version: 5.15.3
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.72.4
lodepng version: 20230410
libzip version: 1.7.3
fontconfig version: 2.13.1
freetype version: 2.11.1
harfbuzz version: 2.7.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof*, input-driver-dbus*, lazy-union*, vertex-object-renderers-indexing*, textmetrics*, import-function*, predictible-output*, python-engine*
Application Path: /tmp/.mount_OpenSCLFgHmD/usr/bin
Documents Path: /home/redkrieg/.local/share
User Documents Path: /home/redkrieg/Documents
Resource Path: /tmp/.mount_OpenSCLFgHmD/usr/share/openscad
User Library Path: /home/redkrieg/.local/share/OpenSCAD/libraries
User Examples Path: /home/redkrieg/.local/share/OpenSCAD/examples
User Config Path: /home/redkrieg/.config/OpenSCAD
Backup Path: /home/redkrieg/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/redkrieg/.local/share/OpenSCAD/libraries
/tmp/.mount_OpenSCLFgHmD/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
[... 445 lines cut ...]
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum

**Additional context**
I don't know what the best solution here is, maybe using a shell script so the first hit at `/tmp/.mount_Open*/usr/bin/openscad-python` would be used?

t-paul, 2025-06-26:
> Instead i use dlopen during runtime to load the python.

Are there any real world applications using this that have a similar deep integration as needed for OpenSCAD? It's probably not too complex when just evaluating python source code 1:1, even though 3 platforms are headache enough.
The current python code uses much more of the API though, is there some sort of change definition for all those functions across a reasonable list of python versions? Binary compatibility is usually much more difficult than source code compatibility on C/C++.

t-paul, 2025-06-26:
> Is there a place where tips and tricks could be collected and shared in the mean time while we all patiently wait for better documentation? Can we create such a place?

Yep, seems like a good idea.

> It seems like github discussions would be a good place for this, but discussions are not enabled for this repo.

Hmm, one option would be the github wiki. But then, there's a number of people who already gave feedback they don't want to create github accounts. I may need to have a look how much work hosting discourse is.

jstebbins, 2025-06-26:
> Hmm, one option would be the github wiki. But then, there's a number of people who already gave feedback they don't want to create github accounts. I may need to have a look how much work hosting [...]

Well, those that don't want to create github accounts could just be lurkers. Those that do can transcribe the collected wisdom of the masses from IRC, GH Issues, etc. I've never used discourse myself. So I don't know if it has a persistent topic based forum where such information could be collected. Happy to use it though if it's the right™ solution for everyone else. But seems like a lot of work if it's only to be used for this one thing.

gsohler, 2025-06-26:
All Python Symbols are already replaced.
I have defined a structure with about 50 (!)  function and data pointers
and the engine is basically running

The big challenges left are
* stdlib path is not calculated reliably, so its not possible right now to
define  custom sys.path locations
* user experience: what is the correct "set" how user select for a python
version