openscad/openscad #6001: documentation suggestion: grammar for comments in .CSG files?

gtoal, 2025-06-26:
The description of csg format in https://github.com/openscad/openscad/wiki/CSG-File-Format doesn't mention the format of comments at all.  I'm guessing that C style '/* ... */' and '//' comments are allowed, but it would be useful to add that info to the description of the file format so that people writing csg files can safely embed comments in them, secure in the knowledge that conforming implementations will read them and not reject the file as containing syntax errors.

kintel, 2025-06-26:
The .csg file format was meant as a way of exporting a strict subset of OpenSCAD, containing only explicit values (no modules, function, variables, included files etc.). As such, OpenSCAD is unlikely to ever export comments.
OpenSCAD will simply import .csg files using the regular OpenSCAD parser, and is thus not sensitive to which exact subset of OpenSCAD is being implemented.

Others may want to implement their own parser, in which case it's a bit more important to know what, exactly, each version of OpenSCAD is capable of exporting.  Human authored .csg is kind of an unexplored territory.

The actual subset is not specified anywhere, and the real answer is "as implemented", supported by existing tests.
Instead of going for a full-fledged spec (which needs to be maintained), perhaps it's possible to define which OpenSCAD subset is implemented in a more terse fashion?

Not sure if this really helped

gtoal, 2025-06-26:
First re Jordan Brown's comment: it's not necessary to document the individual calls (at least as far as a grammar for a parser is concerned) - all <instruction>s use the same syntax.  So as to Marius's reply: yes, all I'm suggesting is a couple of extra lines in the description of the grammar so that parser writers are consistent in reading CSG files and programmers writing them can output code that other utilities can accept.  Documenting the actual primitives is a separate issue and not one I was asking anyone to take on.  (My interest in this came from having just written a utility to manipulate csg files and I wanted to add comments to the output.  https://forum.bambulab.com/t/an-experiment-with-multicolour-openscad/178116?u=gtoal )  Unfortunately comments and whitespace are generally not handled at the grammar level by most grammar descriptions (they're usually handled by a separate line reconstruction phase or a lexer) so it's not even something that merits a BNF description, just a sentence so we know what to expect.

bitbasher, 2025-07-18:
well at the moment import <CSG.csg> is working, but import( file="CSG.csg" ); says it has imported some data but there is nothing displayed

when i paste the .csg content into the editor it does generate correct objects .. and i can export it back out as a new .csg file which can also be included, but not imported

okay .. this file can import .. i even hand edited the last line to add a red cube
`/* test
multmatrix([[1, 0, 0, -24], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
	union() {
		cube(size = [15, 15, 15], center = true);
		sphere($fn = 0, $fa = 12, $fs = 2, r = 10);
	}
}
*/
intersection() {
	cube(size = [15, 15, 15], center = true);
	sphere($fn = 0, $fa = 12, $fs = 2, r = 10);
}
// next is cube
color("red") cube(10);
`

bitbasher, 2025-07-18:
t-paul .. i was responding to kintel's suggestion to check if .csg would handle comments .. i know from my own experiments that include will cope .. 
the import module as written _should_ import .csg files .. it is in the code that a .csg files is accepted for input .. and i see that there is geometry after the import .. no error messages. but nothing displays .. that is a bug no?

t-paul, 2025-07-18:
No, `import()` is not supposed to import .csg, nor .scad. Maybe the error handling can be improved, old style scad scripts tended to ignore most errors and just continue. That has changed over time (and I think for the better). I think more and better error messages are good enhancements, even though not everyone agrees. Bugs, they are not.

Also, no, an existing export does not necessarily mean that must (or can) be a matching import. In most cases that's the case and even desirable, but it's not always possible.

jordanbrown0, 2025-07-19:
The list of types supported by the `import()` module is at https://github.com/openscad/openscad/blob/ef88e36a8e68b54a0080501126fa685a3517eeb7/src/core/ImportNode.cc#L83

It is:  stl, off, dxf, nef3, 3mf, amf, svg, and obj.

Somebody doing an overhaul of the manual erroneously had it state that `import()` supported CSG.  I've removed that statement.

bitbasher, 2025-07-19:
that was me and i made the list by reading it in the code .. or not !
i am sorry .. i somehow got that all wrong. indeed .. import() module will not accept ,csg .. it just fails silently
```
Compile and preview finished.
Total rendering time: 0:00:00.013
 
Parsing design (AST generation)...
Saved backup file: C:/Users/Jeff/Documents/OpenSCAD/backups/test-import-json-backup-sGmQpwkj.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using Manifold...
Geometries in cache: 46
Geometry cache size in bytes: 264032
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.042
Camera:
   Translation: 46.46, 4.11, 3.15
   Rotation:    60.60, 0.00, 6.10
   Distance:    192.04
   FOV:         22.50
Rendering finished.
```

so it looks like it got something from the .csg .. but not enough for Manifold to work with

jordanbrown0, 2025-07-20:
Let's keep this issue for "are comments allowed in CSG" and at most "what are the syntax and semantics of CSG".

bitbasher, 2025-07-20:
can anyone supply a concise and complete spec for what .csg is supposed to do? why is it a format separate from .scad anyway?   Setting that goal determines the semantics .. then we need derive a list of the (minimum set of)  language elements needed to fulfill the goal and we have the syntax too .. so simple .. heh .. so what is the goal of .CSG?

t-paul, 2025-07-20:
I'm not sure it's actually a file format right now, more an unfinished idea.

I believe (I was not there when it was invented, just giving information collected over the years) that it was originally intended to be a communication format between OpenSCAD and slicers which:

1) Can be read without the need to do actual OpenSCAD processing by resolving all functions, variables and modules so we only have a CSG tree of primitives with numbers.
2) Still has things like circles as entity, not converted to mesh.

I have never used any slicer that actually has that implemented, according to the [RepRap wiki](https://reprap.org/wiki/RepRapPro_Slicer) [RepRapPro Slicer](https://github.com/reprappro/host) (last commit 12 years ago) does have the feature.

The only other bigger application I know about is FreeCAD importing CSG (or SCAD by running it through OpenSCAD turning it to CSG).

At some point IceSL could read CSG and I think also SCAD, but I believe they did that by linking to a modified OpenSCAD (and totally ignoring the licence violation along the way). I don't know if they used the CSG tree or the resulting mesh. Later versions used some other means of generating the designs via Lua IIRC.

There are multiple cases where this discussion came up, so there seem to be at least a couple of instances where people want to use CSG in addition.

Side note: If we ever manage to actually give built-in modules (like `linear_extrude()`) features that can handle lambdas, the whole CSG idea might be getting tricky to impossible.

jordanbrown0, 2025-07-21:
I think it's also a mechanism for testing and diagnosing OpenSCAD itself and OpenSCAD programs.  If you are confused by what the previewer is showing you, looking at it in CSG will let you see it without loops, variables, et cetera, just raw geometry.  For OpenSCAD itself, for instance, it helps to verify that argument processing has yielded the right results.