openscad/openscad #6005: Create Warnings with the assert function

Nexusnui, 2025-06-28:
**Is your feature request related to a problem? Please describe.**
I would like an option to issue warnings with the assert function, instead of a full error, that stops rendering.

**Describe the solution you'd like**
New parameter in the assert function, that enables a warning mode. The default should result in the current behaviour. This new parameter should probably be a boolean, unless there is a reason for additional modes.
Alternatively there could be a new function for warnings.

**Describe alternatives you've considered**
Use Echo to show warning, but this is not very visible in the console. Additionaly users of a library might not expect them and they do not appear in the Error-Log.

jimb15122, 2025-11-14:
Another option, that requires no new syntax, would be to simply honor current color() for echo to set the text or bk color.   But this would affect existing echo output.  Color coding echo's could have real utility.  Perhaps a config setting to enable it.  or how about. #echo(). !!

Nexusnui, 2025-11-14:
You can also click on the warning to jump to the source.

jordanbrown0, 2025-11-14:
The echo variant is a non-starter, because that's already a valid "echo" invocation.  It's not really possible to add functionality to echo(), because all positional and named arguments already have well-defined semantics.  We should perhaps add a new output module that is more extensible.

I would say that new semantics for ! or # are also a non-starter.  There's just no justification for that level of semantic change, when there are options that are straightforward module invocations.

My inclination is for `warn()` to be echo-like, but there are arguments for something like the other two variations.