openscad/openscad #6008: Defining the result of undefined operations.

gtoal, 2025-06-30:
Am I correct in thinking that the colour of:
```
intersection() {
  color("red") cube([1,1,1]);
  translate([0.5,0.5,0.5]) color("green") cube([1,1,1]);
}
```
is undefined or at least implementation-dependent?

If the outcome is not already defined, I'ld like to suggest that the colour of the result, when differently-coloured objects are intersected, is set to be that of the first object.

This would make the operation similar to 'difference' (where the result clearly has to be the colour of the first object) and creating a consistency across all operations (when doing so is logical and not an upheaval to the code base or existing models) should make the rule easier to remember for users.

I suspect the same logic could be applied to the intersected common parts in a 'union' result as well.  I'm not sure if there are any other operations where a choice has to be made -  union, lazy union, and group should of course preserve the original colours of the non-intersecting components.

jordanbrown0, 2025-06-30:
In any visible face, it came from one object or the other.  It gets the color of that object.

Thus:
```
intersection() {
    color("green") cube(15, center=true);
    color("red") sphere(10);
}
```
yields:

[image]

And while you might think that
```
difference() {
    color("green") cube(10);
    color("red") sphere(10);
}
```
would be all green, you'd be wrong:

[image]

For either case, if there's a coincident face, a face that's the same for both objects, the color is undefined and you get Z-fighting.

Historically, color is supported only in the previewer, and the previewer uses black magic to do its work.  It does not do *anything* with volumes; it operates only on faces.

We now have some color support in rendering, but it's still face-based.

There is an argument (which I happen to agree with) that we should move to volumetric color for rendering, which would yield results like you suggest.  But we're not there now.  (And why does anybody care about the color of the inside of an object?  Because when you lay down filament, it matters - especially if the filament is different materials, not just different colors.)

This tension, and this expectation of possible evolution, is why the behavior of color has never been fully documented.

closed 2025-07-02 by kintel

gtoal, 2025-07-02:
I appreciate the explanation of why it is the way it is - it does make the apparently random results more predictable - but what you're saying is that the current results are not the result of a deliberate intended behaviour but rather an unwelcome (if not unintended) side-effect of the specific implementation; that the user's expectation is really that when you assign a colour to an object, the entire object would be that colour rather than just the surface...  which is reinforced by the fact that the result does not propogate to the final 3D print (because the coloured face has 0 depth).

So what I'm suggesting is primarily just a documentation change, to define the _intention_ as being what the user intuitively expects, rather than what the implementation happens to display in the preview window.  There's nothing wrong with saying to the users that the intention of an intersection operation is that the resulting object has the internal colour of the first component even if the rendering currently happens to display some of the surfaces differently.

That said, actually changing the code, even with the current implementation as you've described of it being an operation on surfaces rather than volumes, may only need a trivial change and not the complex rewrite that I think you might be wary of: here's how I would do it:

When your input is:

```
  intersection() {
    color("red") cube([1,1,1]);
    translate([0.5,0.5,0.5]) color("green") cube([1,1,1]);
  }
```

you could make the results equivalent to an operation that is defined volumetrically by hoisting any color() node attached to the first parameter up to apply to the result of the entire intersection operation, i.e. treating the intersection operation as if it had been generated by:

```
  color ("red") intersection() {
    cube([1,1,1]);
    translate([0.5,0.5,0.5]) cube([1,1,1]);
  }
```

(and indeed just adding the color() node above the expression without even removing any color() nodes around the parameters would work just as well and be simpler to implement:

```
  color("red") intersection() {
    color("red") cube([1,1,1]);
    translate([0.5,0.5,0.5]) color("green") cube([1,1,1]);
  }
```
)

which ought to be a simple manipulation of the internal CSG tree generated by the parser and could probably even be done *by* the parser as it generates the internal data structure corresponding to the expression before it's passed to the CSG logic.  (Of course, the user could equally write their source code like that explicitly, and I probably will, but it would be nice to have OpenSCAD do it and align the implementation with the intention).

I do realise that if the first parameter of the intersection command is a multi-coloured object and is not assigned an explicit colour, the suggestion above will have no effect and the current behaviour will still be observed.  The suggestion would not be a complete fix, only an incremental improvement.  I'ld be OK with that; you might not be.  But I'm primarily proposing a _documentation_ change to say that the ultimate _intention_ is to treat colours as volumetric attributes instead of surface attributes, rather than asking for an _implementation_ change.

Thanks for taking the time to explain the current implementation.

Graham

gtoal, 2025-07-02:
At the risk of drifting a little off-topic here (I apologise), on the subject of: "For either case, if there's a coincident face, a face that's the same for both objects, the color is undefined and you get Z-fighting." ...

This is a common problem in computer science and affects a lot of other fields than just 3D modelling; there's a parallel issue in raster graphics for example which you might have noticed, when a rectangle is rendered by drawing two adjacent triangles and the implementation gets it wrong, with the join between the two diagonals being visible because the two diagonals don't quite align exactly!

The standard solution is that lines, areas and indeed volumes should be defined as semi-open intervals, and for consistency, the rule of thumb has always been: 'lower bound inclusive, upper bound exclusive'.  If you apply this to CSG operations, you don't get Z-fighting, but you do get different (but consistently reproducible) colours on the near/left/lower faces from those on the far/right/upper faces -- which is somewhat unintuitive to the majority of people, who didn't have this rule of thumb drummed into them in their CS classes!

jordanbrown0, 2025-07-02:
No, there's no `material()` operator.

With respect to Z-fighting... not my area of interest/expertise.  If you have ideas for improvement, you might take a look at the current implementation and/or file a separate issue.

gtoal, 2025-07-02:
@jordanbrown0:
> No, there's no `material()` operator.

I think I've been reading too much too quickly!  I'd seen mentions of it in issues here and when skimming https://tams.informatik.uni-hamburg.de/publications/2023/BSc_Nicolaus_Johannes_Leopold.pdf and didn't realise it was only a suggestion and not something already implemented in the nightlies!  My mistake... (It probably should be implemented though, and indeed generalised to allow any arbitrary attribute that a filament can have, possibly supplemented by a user-defined function to say whether user-defined attributes of two different filaments should forbid them from being unioned together!)

> With respect to Z-fighting... not my area of interest/expertise. If you have ideas for improvement, you might take a look at the current implementation and/or file a separate issue.

That wasn't a suggestion, just a wry comment on how even doing something the most consistent way isn't always best from the point of view of the user's expectations!  What is it they say? "Foolish consistency is the hobgoblin of little minds" :-)

@kintel:

> There has been long discussions in the past on how to properly define, design and implement volumetric colors. The most comprehensive discussion may be https://github.com/openscad/openscad/issues/1608.

Thanks for the pointer; also I've just now found your forum/mailing list and subscribed, so I won't drag out this entry in the github issues now that I know there's a better place for discussions!  Thank you for putting up with my newbie questions.