openscad/openscad #6015: Support for OpenType features in code editor

marcuserronius, 2025-07-08:
I've been using [GitHub's "Monaspace" fonts](https://monaspace.githubnext.com/) for my coding for a while now. It's not just a nice font; it uses OpenType features like ligatures, contextual alternates, and stylistic sets to make code more readable. Unfortunately, I can't seem to get it to work properly in the OpenSCAD editor. The stable version doesn't display any of the stylistic alternates, even when using the "Frozen" versions of the fonts, which are supposed to force it to work in incompatible editors. The Nightly I have installed (2025.07.06) DOES work with the Frozen versions, but it causes odd things to happen in the editor when you are editing the characters that make up the ligatures; the caret moves to non-fixed-width positions, and the ligatures themselves change shape and size when the caret is inside them.

I'd like to see options in the editor settings for turning on or off the various OpenType features that a font might support. I imagine this as either a checkbox for each feature, or a more compact text field where you can type in the names of the features you want enabled. Both of these options are used in other text editors. Adding this would allow the editor to display the font as intended and allow the user to turn off any features they don't like, as intended by the makers of the font.

I'm using the Frozen version right now as a partial workaround, but there are definitely some bugs in how the features are handled, as described above. The good still outweighs the bad for me, but it would be really nice to have it work correctly.

Caveat: I'm not sure how well QT supports OpenType features. What info I could find suggested this should be possible, but I'm definitely not a QT app developer. It may be that enabling the features and/or fixing the display bugs is not possible, or is more trouble than it's worth.

jordanbrown0, 2025-07-08:
The editor is QScintilla, with some custom configuration.  There's a decent chance that support would need to be there or in Qt.

kintel, 2025-07-08:
Which OS are you using and which version of Qt show up in Help->Library Info ?

marcuserronius, 2025-07-08:
@jordanbrown0 : I'll take a look at that
@kintel : macOS 15.5.0, Qt 6.5.5, QScintilla 2.14.1

marcuserronius, 2025-07-08:
An uninformed browsing of the Qt docs suggests that control of stylistic sets definitely isn't supported before Qt 6.7, but might be supported in 6.7 and up? Stylistic sets aren't mentioned directly, but the method of turning on an arbitrary feature is, and stylistic sets are an arbitrary feature of that type, so... maybe?

I don't know what's involved in moving up to Qt 6.7 or higher, either, but I imagine it's not an insignificant project.

kintel, 2025-07-08:
We'd have to move to 6.8 LTS I think - it might be good enough to use by now, but it's always a bit of work to iron out any UI corner cases.