openscad/openscad #6020: Please add the time at which a compilation is begun at the top of the Log

WillAdams, 2025-07-12:
When compiling a model which takes a long time, folks will often switch away from the window, leaving the application running in the background --- if the compile takes a really long time, one has to note when it was begun so as to keep track of how long it is taking.

If OpenSCAD would note the time in the log, so instead of:

>Saved backup file: C:/Users/willa/OneDrive/Documents/OpenSCAD/backups/gcpgc-backup-NEjSyZqU.py
>Running Python 3.11.5 without venv.
>Compiling design (CSG Tree generation)...
>Rendering Polygon Mesh using Manifold...

one had something like to:

>Saved backup file: C:/Users/willa/OneDrive/Documents/OpenSCAD/backups/gcpgc-backup-NEjSyZqU.py
>Running Python 3.11.5 without venv, starting at: 07:49 AM 7/12/2025
>Compiling design (CSG Tree generation)...
>Rendering Polygon Mesh using Manifold...

then when one switched to the window it would be obvious when the process was begun.

I've been scribbling down begin times either in a text file or on paper, but it's a nuisance.

kintel, 2025-07-12:
This makes sense to me. Should be pretty easy to implement if someone wants to make a shot at it.

jordanbrown0, 2025-07-12:
I'd say that timestamping everything (or at least everything other than echo()) is more likely, so I'm calling this a duplicate of #3570.

closed 2025-07-12 by jordanbrown0