openscad/openscad #6027: OpenSCAD doesn't bring window to front or switch spaces on "Next Window" or "Previous Window" menu selection

Zenmaster43, 2025-07-16:
**Describe the bug**
The "Window" menu items that select a window, ie. Next Window, Previous Window do not bring the active window into view on MacOS Sequoia 15.1 or earlier versions. The same is true when switching from another open application back to OpenSCAD. I have to actually slide three fingers up to miniaturize all open windows on the space, select the appropriate space from the mission control menubar and then select the miniature OpenSCAD window to bring it to front.

**To Reproduce**
Steps to reproduce the behavior:
1. Lauch OpenSCAD
2. Select "New File" or "Open File..." and pick a file.
3. Hold down the command key and press the tab key to select another open application that has its active window open in another space and release the keys. The selected application's active window appears in front of the containing space's desktop.
4. Hold down the command key and press the tab key repeatedly as needed to select the OpenSCAD icon and release the keys. The OpenSCAD menubar appears, but the space is not switched back to where the OpenSCAD window is at.
5. Choose "File New" from the "File" menu. Nothing changes visually. (The new file window is opened in the space that contains the first OpenSCAD window, currently not visible. Switching spaces back to the one containing the first OpenSCAD window reveals that the new file window was created but its containing space was not automatically brought into view).

**Expected behavior**
Step 4: the opened OpenSCAD editor should reappear along with the OpenSCAD menubar.
Step 5: The space containing the new OpenSCAD window should be brought into view. 

Also:
1. Launch OpenSCAD.
2. Choose "New File" in the "File" menu, twice, to create two OpenSCAD windows.
3. Choose "Next Window" or "Previous Window" in OpenSCAD's "Window" menu. Nothing happens.

**Expected behavior**
In step 3, the first OpenSCAD window should come to the front.

**Code reproducing the issue**
(No code required.)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info:
 - OS: MacOS 15.1 Sequoia
 - System: MacBook Air M2
 - OpenSCAD Version 2024.11.10 (git 6f5932feb)

Zenmaster43, 2025-07-16:
This is a persistent problem that wastes a lot of my time. Has anyone else experienced this? It has been the same over the past several versions of MacOS and OpenSCAD, for as far back as I can remember.

kintel, 2025-07-16:
Could you verify that this is still an issue with 2025.07.*?

You're still on 2024.11.10, which is a bit old by now.

jordanbrown0, 2026-08-15:
While I understand the sentiment, it doesn't add value to the conversation.  Hidden as off-topic.