openscad/openscad #6031: import dxf fails as function with import-function enabled, while calling import() as a module works

bitbasher, 2025-07-18:
**Describe the bug**
dev snapshot 2025,3,31 onh Windows 11
i have enabled the import-function option in the prefs.

`info=import(file = "example009.dxf", layer = "plate");
echo(info);

import(file = "example009.dxf", layer = "plate");
`

running that script gives me:

`[WARNING: variable "layer" not specified as parameter in file fail-import-dxf.scad, line 1](https://github.com/openscad/openscad/1,D:/repositories/fail-import-dxf.scad)
ECHO: 999
`
so that error message is .. just wrong. the layer param is right there .. or is there a function named import() that does not have the same argument list as the module inport() .. which would make no sense.

the second call to import() as a module works

[image]

**Expected behavior**
i expect both calls to work the same way

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - Intel PC 64-bit
 - OpenSCAD 2025.3.31 downloaded from the web site

jordanbrown0, 2025-07-18:
Not exactly a duplicate.  #6030 says "should have rejected the DXF", while this one says "should have done something useful with the DXF".

With regard to this one, at the moment it's working-as-designed.  import-the-function's only capability is to import JSON.  It might well be useful and appropriate to give it the ability to read other file types, but so far nobody has done that.

t-paul, 2025-07-18:
> That message could certainly be better.

True, but that applies to all functions and modules, not just specifically to `import()`. With a bit of extension to the built-in docs, it could (optionally?) show the calling convention for the function a bit like the editor tooltip.

t-paul, 2025-07-18:
That's a reasonable feature request, but not what the issue describes. To reduce noise, it would be better to have a new issue with a clear description of the feature request including at least *some* hint of what it should return. And I would also exclude a generic request to return "just all geometry as data".