openscad/openscad #6043: Make Value a double ...

pkriens, 2025-07-25:
The current Value object is 24 bytes. Since most objects will be doubles, most Value objects will waste 16 bytes. 

The extra space that is needed adds up in memory consumption but also performance due to cache locality. It will also require conversions to support SIMD processing. The reason the Value needs shared_ptr's is because it maintains a shared reference to larger objects like strings, objects, arrays, or functions. 

Javascript, and I think Lua, provide a similar abstraction as Value. However, they only use 8 bytes per Value: a double. This is possible because a double (8 bytes) reserves 52 bits for Not a Numbers (NaN). By using the NaN space in a double it is possible to maintain references to shared pointers. That is, if a Value is a NaN, the 52 bits are extracted and used to locate a struct that maintains the reference count and a pointer to the actual type. If the reference count goes to zero, the pointer is deleted.

Operations would still require a dispatch on the underlying types of the values but this can be done very efficiently. 

Is this idea in principle welcome or should this part of OpenSCAD be left alone? 

If there is some enthusiasm, I can make a prototype.

pca006132, 2025-08-03:
I don't think this affects performance. IMO the dynamic dispatch and name lookup in the interpreter is much slower. Ideally we should translate to some bytecode for more efficient execution.

t-paul, 2025-08-03:
I guess it mainly impacts the depth of recursion when tail-recursion is not available. So any smaller Value would be nice, but I'm unsure if that's a worth the possible gains at the current state of things overall.

jordanbrown0, 2025-08-03:
It doesn't seem like this scheme would improve *performance* per se.  In fact, it seems like it would slightly hurt performance for everything but numbers, since they would require an extra indirection.  What it would dramatically help is memory consumption for large arrays of numbers, which would perhaps see their memory consumption divided by three.

---

That said, I'm kind of nervous about depending on the details of the representation of doubles.

pkriens, 2025-08-07:
> I don't think this affects performance. 
> It doesn't seem like this scheme would improve performance per se.

Memory usage has a very large impact on CPU cache locality which is crucial for performance when you need a lot of data in loops, which is typical for large 2d/3d path processing for example. I can also see some optimizations for these vectors consisting of doubles only.

However, the current code of the language implementation is not very optimized so overall performance will be limited. I see a number of potential optimizations.

> Ideally we should translate to some bytecode for more efficient execution.

Yes, but I am willing to work on this, changing the whole language might then be the next step. (WASM is interesting!)

> ... any smaller Value would be nice, but I'm unsure if that's a worth the possible gains at the current state of things overall.

It will not cost you anything  to let me try it. Clearly if I do the work and there are no improvements we skip it. Just want to be sure I am not running in some religious objections once I did the work.

>  In fact, it seems like it would slightly hurt performance for everything but numbers, since they would require an extra indirection.

Since the variant handling will go away, I'd bet you $10 that also the indirect types will also be faster :-) However, I doubt this is noticeable with the current codebase.

> That said, I'm kind of nervous about depending on the details of the representation of doubles.

The NaN model & bits for doubles are highly standardized in IEEE 754-2008 and supported by all processors with floating point. It would be suicide for a processor to try to introduce a new format and not provide backward compatibility. Also, the same mechanism is used in Lua and Javascript. I've actively used this extensively in a Java framework. It is as standard as it gets in our world. Of course I could add a fallback but looking at the CPU landscape that seems quite unnecessary unless we want to run OpenSCAD on an ESP32? :-)

pkriens, 2025-08-08:
I did look at that and an idea might be to use WASM. There are WASM libraries that will do JIT for all popular platforms. Compiling OpenSCAD source code to WASM and then running it in a WASM container will make it lightingly fast. I then looked for a runtime library compiled to WASM that implemented common types like string, number, list, etc but couldn't find one except maybe Javascript itself. It is not rocket science to make the runtime since you can just compile C++ code to WASM bytecodes. I think WASM is about 10Mb without runtime support.
Other alternative is to include a minimal Java VM  ... if that would be an option I'd do the work ... I am a [world expert on bytecodes and class handling in Java](https://github.com/bndtools/bnd) ... but this would be 25-30 Mb with the hotspot JIT with a bare minimum VM. Hotspot is lightingly fast, it optimizes on the fly using runtime behavior. Would definitely provide some cool possibilities.

In both cases security, the issue we talked about before, could easily be handled. 

However, that is a major effort  and will almost certainly show different behavior from today's code, and then I mean not just the performance.

I am talk to Adriano to see if I can find out where the time goes. One thing I noticed is that for_each is taking a lot of time in my favorite function. The function setup and code analysis to handle tail recursion was a significant amount of time. Maybe caching the function setup would be a big win. Who knows.