openscad/openscad #6045: OpenSCAD reproducible hang on Apple Silicon involving calls to removed/renamed functions

dgatwood, 2025-07-25:
**Describe the bug**
While working on models, I keep running into situations where I make a partial change to rename a function and miss some reference to it, and OpenSCAD locks up when I save (because of the re-rendering).  This has happened several times recently.

I made a copy of one of those bad files so I could file a bug, so I was able to create a trimmed-down edge case.

I think all the failing cases may involve rotate_extrude in some way, some directly, some indirectly, but not certain.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the attached .scad file.
2. Save.

**Expected behavior**
No hang, just errors.

**Code reproducing the issue**

```
// https://github.com/rcolyer/threads-scad
use <threads-scad/threads.scad>
                
function inches_to_mm(inches) = inches * 25.4;
                        
module piston_valve_casing(bore=0.413, odd=true, valve_thread_pitch = 2) {
    casing_height = 101.5;

    mmbore = inches_to_mm(bore);
    AugerHole(24, 22, 5, valve_thread_pitch, position = [0, 0, casing_height - 5], rotation = [0, 0, 180]) {
        AugerHole(24, 22, 5, valve_thread_pitch) {
            color([1.0, 0, 1.0]) {
                difference() {
                    translate([0, 0, 0]) cylinder(casing_height, 13, 13, $fn=256);
                    translate([0, 0, -1]) cylinder(casing_height + 2, 11.01, 11.01, $fn=256);
        
                    // Allow 10mm at the bottom for the compressed spring and plug; shift up the tubing holes
                    translate([0, 0, 10]) {
                        rotate([0, 90, 45]) translate([odd ? -46 : -30, 0, -18])
                            cylinder(10, mmbore/2, mmbore/2, $fn=256);
                        rotate([0, 90, 45]) translate([odd ? -30 : -46, 0, 9]) 
                            cylinder(10, mmbore/2, mmbore/2, $fn=256);
                        rotate([0, 90, 135]) translate([-46, 0, -18])
                            cylinder(10, mmbore/2, mmbore/2, $fn=256);
                        rotate([0, 90, 315]) translate([-30, 0, 9])
                            cylinder(10, mmbore/2, mmbore/2, $fn=256);
                    }
                }

                // Flat side.
                rotate([0, 0, 45]) translate([-6.5, 9.02, 0]) cube([13, 1, casing_height]);
                rotate([0, 0, 45]) translate([-5, 9.02, 0]) cube([10, 2, casing_height]);
                rotate([0, 0, 45]) translate([-5, 9.02, 8]) cube([10, 2.8, casing_height - 16]);

                if (false) {
                    // Bottom threads
                    difference() {
                        ScrewThread(28, 8, pitch=valve_thread_pitch);
                        translate([0, 0, -1]) cylinder(12, 11.01, 11.01, $fn=256);
                    }

                    translate([0, 0, casing_height - 8]) difference() {
                        translate([0, 0, 8]) rotate([0, 180, 0]) ScrewThread(28, 8, pitch=valve_thread_pitch);
                        translate([0, 0, -1]) cylinder(12, 11.01, 11.01, $fn=256);
                    }
                }
            }
        }
    }
}

// This line causes the crash.
        translate([0, 0, 2]) ScrewThread(24, 5, pitch=valve_thread_pitch);  // 24, 22,

```

**Environment and Version info (please complete the following information):**
 - OS: macOS 14 (Sonoma)
 - System: Apple Silicon (M1 Max)
 - OpenSCAD Version: 2025.07.20 (git 721d8bc0e) downloaded from GitHub

**Library & Graphics card information**
Can't seem to copy from that window (bug).

**Additional context**
The one required libraries (from github) is listed at the top.

jordanbrown0, 2025-07-25:
A hang is absolutely wrong.

Properly fails for me on Windows with a pretty much current build.
```
WARNING: Ignoring unknown variable "valve_thread_pitch" in file ., line 53
TRACE: called by 'ScrewThread' in file ., line 53
TRACE: called by 'translate' in file ., line 53
Execution aborted
```

jordanbrown0, 2025-07-25:
Sorry, to be absolutely clear:  I agree that a hang is a bug.  I do not duplicate that bug.

jordanbrown0, 2025-07-26:
Also:  it would be good to reduce this to a minimal demonstration.  Unfortunately, with a non-trivial library that can be tough, but still:  merge in the library to get rid of the use<>, then comment things out until removing anything causes the problem to not happen.

pca006132, 2025-08-03:
> A hang is absolutely wrong.
> 
> Properly fails for me on Windows with a pretty much current build.
> 
> ```
> WARNING: Ignoring unknown variable "valve_thread_pitch" in file ., line 53
> TRACE: called by 'ScrewThread' in file ., line 53
> TRACE: called by 'translate' in file ., line 53
> Execution aborted
> ```

Crash can manifest as hang in some systems.

pca006132, 2025-08-04:
But still, it seems that it is just infinite recursion even with your updated example. If you do it through the command line:
```
openscad hang.scad -o hang.stl
```