openscad/openscad #6046: render() module fails silently in some conditions of scope

bitbasher, 2025-07-26:
**Describe the bug**
the render() module is supposed to tessellate shapes without regards to render mode, but it fails silently when there are too few objects in its scope. 
**To Reproduce**
run this
```
translate([-5,0,0]) color("red") sphere(2);
color("green") sphere(2);
translate([5,0,0]) color("blue") sphere(2);
```
as expected the three spheres are not tessellated:
[image]

then this:
```
render( convexity=3 )
translate([-5,0,0]) color("red") sphere(2);
color("green") sphere(2);
translate([5,0,0]) color("blue") sphere(2);
```
the result is the same - all three spheres are not tessellated

and again:
```
render( convexity=3 ) {
translate([-5,0,0]) color("red") sphere(2);
}
color("green") sphere(2);
translate([5,0,0]) color("blue") sphere(2);
```
again, the same result

but include two objects in the {} scope 
```
render( convexity=3 ) {
translate([-5,0,0]) color("red") sphere(2);
color("green") sphere(2);
}
translate([5,0,0]) color("blue") sphere(2);
```
now two of the three are tessellated .. which is correct
[image]

extending it further
```
render( convexity=3 ) {
translate([-5,0,0]) color("red") sphere(2);
color("green") sphere(2);
translate([5,0,0]) color("blue") sphere(2);
}
```
now all three are tessellated:
[image]

**Expected behavior**
i expected trial 2 would show the red sphere tessellated 
that trial 3 would show the same

the other trials worked correctly, this report is about the apparent issue of render() failing to work when it had only a single child to in its scope.

OpenSCAD  2025.07.20

**Environment and Version info (please complete the following information):**
 - hardware
Processor	Intel(R) Core(TM) i9-14900K   3.20 GHz
Installed RAM	32,0 GB (31,8 GB usable)
Device ID	287FD8FA-86F8-4F21-83CA-3A4CE04FF4A1
Product ID	00330-53538-17717-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
- windows
- Edition	Windows 11 Pro
Version	24H2
Installed on	‎18/‎05/‎2025
OS build	26100.4652
Experience	Windows Feature Experience Pack 1000.26100.128.0

 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

t-paul, 2025-07-26:
I don't think that's a bug. `render()` does not guarantee a tessellated mesh, it only forces the result is a single mesh and not a CSG tree. So if it's input is a primitive using faces with more than 3 vertices it's still just a NOP.

If you put 2 primitives into the scope after `render()` this gets processed as union through manifold, hence you get the tessellation effect. So basically the union via Manifold causes the tessellation, it has nothing to do with the `render()`.

jordanbrown0, 2025-07-28:
Generally speaking, a render (whether F6 or render()) does nothing if the only thing it's operating on is a single primitive. and no boolean operations are required.  (In fact, it's even "smart" enough to do nothing if the other operand of a boolean operation is empty.)

There's a good argument that it would be better if it went through the motions anyway, so that incorrect polyhedra (including imported) are reported in all cases.  The performance cost would be trivial, and it would save a lot of questions.

t-paul, 2025-07-28:
Also, just noting you are always using `convexity = 3`. You know this has nothing to do with the result of the `render()` either? This parameter is purely a hint for preview display and totally unrelated to the mesh operations.

t-paul, 2025-07-28:
No, it has only one, the one described above: Turn everything it gets into mesh (without any guaranties regarding a specific tessellation). If it's already a mesh, it has nothing to do.

Has square root multiple modes of operation? `sqrt(4) = 2`, the input 4 changed to output 2. `sqrt(1) = 1`, the input is unchanged so it does nothing.

Why the focus on triangles? There's no definition whatsoever how the operations work. The primitives have a defined mesh structure depending on the resolution settings. Anything beyond is not specified.

jordanbrown0, 2025-07-28:
Like Torsten said, nobody ever said that the mesh would only consist of triangles.  (And:  even if it only consists of triangles, there's no guarantee of which triangulation it will choose to break up a larger polygon.)  Just because you can observe a particular behavior does not mean that that's the only possible and legal behavior.

With respect to implicit union... Almost all OpenSCAD operations implicitly union their children.  Offhand, the only exceptions are difference, intersection, and minkowski.  And the top level is always unioned.  (Lazy union excepted, but as has been discussed lazy union is not ready for prime time.)

When you have three spheres that don't touch each other, and you render() them or F6 render them, the result is one mesh with two disjoint pieces - what some tools will call two volumes.  (CGAL render will call this four volumes:  each sphere, plus the rest of the universe.)

pca006132, 2025-07-29:
> but that is what happens visually

But this is not the contract. A mesh can be a polygonal mesh or triangular mesh, and there can be multiple ways to tessellate it. The implementation chooses some way depending in what it sees fit. Eitherway, it is converting csg tree to a single mesh.

nophead, 2025-07-29:
Without render() the preview does differences, etc using pixel based graphics operations. With render() the geometry is converted to a mesh and preview simply draws that mesh. I find that very useful because it can speed up preview massively if you have something like a nut that there are hundreds of in an assembly.

I think it also results in cached mesh so it only gets done once when there are many instances.