openscad/openscad #6050: SHA256 for the Windows Installer does not match.

SarahDsll, 2025-07-30:
I have downloaded the zip and exe for 64 bit Window and the sha256 values of both do not match the stored values.

C:\ > Get-FileHash "C:\Users\Downloads\OpenSCAD-2021.01-x86-64-Installer.exe"

Algorithm       Hash                                                                                                         
---------       ----                                                             
SHA256          E3B0C44298FC1C149AFBF4C8996FB92427AE41E4649B934CA495991B7852B855       

C.\...\Downloads> Get-Content .\OpenSCAD-2021.01-x86-64-Installer.exe.sha256
fbe1e590e1af2af863f5d92bd263eb4644d0ee29352e3ab566366e87a8f6ed12

t-paul, 2025-07-30:
I just tried on Linux with `sha256sum` and it matches. 

```
$ wget https://files.openscad.org/OpenSCAD-2021.01-x86-64-Installer.exe https://files.openscad.org/OpenSCAD-2021.01-x86-64-Installer.exe.sha256 
--2025-07-31 00:29:21--  https://files.openscad.org/OpenSCAD-2021.01-x86-64-Installer.exe
Resolving files.openscad.org (files.openscad.org)... 45.129.182.88
Connecting to files.openscad.org (files.openscad.org)|45.129.182.88|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 21876824 (21M) [application/octet-stream]
Saving to: ‘OpenSCAD-2021.01-x86-64-Installer.exe’

OpenSCAD-2021.01-x86-64-Insta 100%[=================================================>]  20,86M  12,8MB/s    in 1,6s    

2025-07-31 00:29:23 (12,8 MB/s) - ‘OpenSCAD-2021.01-x86-64-Installer.exe’ saved [21876824/21876824]

--2025-07-31 00:29:23--  https://files.openscad.org/OpenSCAD-2021.01-x86-64-Installer.exe.sha256
Reusing existing connection to files.openscad.org:443.
HTTP request sent, awaiting response... 200 OK
Length: 113 [application/octet-stream]
Saving to: ‘OpenSCAD-2021.01-x86-64-Installer.exe.sha256’

OpenSCAD-2021.01-x86-64-Insta 100%[=================================================>]     113  --.-KB/s    in 0s      

2025-07-31 00:29:23 (118 MB/s) - ‘OpenSCAD-2021.01-x86-64-Installer.exe.sha256’ saved [113/113]

FINISHED --2025-07-31 00:29:23--
Total wall clock time: 1,8s
Downloaded: 2 files, 21M in 1,6s (12,8 MB/s)

$ cat OpenSCAD-2021.01-x86-64-Installer.exe.sha256
fbe1e590e1af2af863f5d92bd263eb4644d0ee29352e3ab566366e87a8f6ed12  releases/OpenSCAD-2021.01-x86-64-Installer.exe
$ sha256sum OpenSCAD-2021.01-x86-64-Installer.exe
fbe1e590e1af2af863f5d92bd263eb4644d0ee29352e3ab566366e87a8f6ed12  OpenSCAD-2021.01-x86-64-Installer.exe
```

I found some reference that e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855 might be the result for an empty file. Maybe check if your browser just created a 0 length file and blocked the download, I've seen cases where it requires another confirmation in the download dropdown list to actually download the file.

SarahDsll, 2025-07-31:
Thanks for your answer, it was actually because the download was partially blocked.

closed 2025-07-31 by SarahDsll