openscad/openscad #6054: Wireframe Xray view

lorenuars19, 2025-07-30:
View option to draw wireframes on top of everything

maybe dim the color of edge that are behind a surface or something

gtoal, 2025-08-01:
I just now saw in the emacs mode for scad that it can invoke openscad with the parameters `--view=` and one of `axes`, `crosshairs`, `edges`, `scales`, or `wireframe` as the parameter to view.  So maybe it already supports the option you're asking for?

jordanbrown0, 2026-10-30:
Just to record some thoughts:

I was just thinking about wireframes for some reason, and wondered why we couldn't easily reimplement wireframe mode, by drawing lines along all of the edges in the mesh.  The answer is that because the mesh is triangulated, the result wouldn't be a wireframe:  non-triangular faces would be split into potentially many triangles, each with their own lines, rather than just having a single line around the face.

It seems like one could do something by checking, for each edge, whether the adjoining triangular is coplanar with this one.  But that would require some fast way to find that adjoining triangle, and for a triangle-soup mesh seems only well-defined for manifold meshes.

jordanbrown0, 2026-10-30:
To answer @gtoal's comment:  there used to be a wireframe mode, but for reasons that I don't know (not to imply that they aren't good reasons; I just don't know them) it got removed.  (I think it might have been that it depended on CGAL behavior not present in Manifold.)

gtoal, 2026-11-13:
Funny you should mention that, I was chatting about planar mesh simplification recently with a fellow coder for the 1980's Vectrex game console (where all graphics are wireframe).  I found out that CGAL actually does have a package for this: https://doc.cgal.org/latest/Surface_mesh_simplification/index.html (I found out about that long after I'd prototyped a simple demo version myself for our Vectrex gaming code, to convert simple triangulated spaceship models to even simpler ones that draw better... https://gtoal.com/vectrex/mesh/mesh.c - may be too simple for OpenSCAD use but good enough to explain the process I guess.)