openscad/openscad #6055: beautify workflow

AaronVerDow, 2025-07-31:
I would like to have beautify applied uniformly and enforced to make it easier to submit clean pull requests.

I created a quick GitHub worker that can run uncrustify, but there is the issue of matching uncrustify versions between the worker and the local development environment.

As a solution to this, I am looking at adding the uncrustify source as a submodule. It does not have any extra dependencies on top of what OpenSCAD already requires, and it compiles relatively quickly. This would allow a specific version to be picked for the project, and it would be applied uniformly across development environments and the GitHub worker.

Would the dev community be interested in this change?

This would be a new worker that rejects pull requests if beautify finds any errors along with a local script that can fix the errors using the exact same version of uncrustify.

AaronVerDow, 2025-07-31:
I'm not familiar with OBS, but from what I'm seeing I don't see a clean path for local development. Installing a package from a custom source is a higher bar than I'd like to require.

I'm good with Docker, I just leaned away from it to avoid the extra local development requirements. However, with Docker, `act` can be used with the workflows.

I just had another idea: do not add a submodule, just have the script git clone a specific commit of uncrustify and compile that as needed. The commit version can be coded into the script to ensure it's the same for everyone.

t-paul, 2025-08-02:
Right, I was only considering the github action / CI part where we have (almost) full control over the environment. When trying to include the local development, things get a bit more difficult.

Another thing to consider is if maybe going with something else than `uncrustify` is an option. IIRC at the time the script was introduced, `clang-tidy` had some issues getting to the results expected. As that was quite some time ago, that problem may be gone, or maybe there are even other tools.

nomike, 2025-08-05:
I was just filing a ticket about exactly that topic :-).

I'm mainly developing Python code at work and personally, but I recently started taking over package maintenance for the openscad and pythonscad package for the [GNU Guix package manager](https://guix.gnu.org). I'm also starting to get more and more involved with PythonSCAD, having the goal get the build system more stable and adding packages to more distributions. I plan to contribute any improvements back to the OpenSCAD project, once they are complete and stable enough.

I have a few cents which I would like to add here:

1. GNU Guix is a functional programming package manager (derived from Nix) and it is preferred wherever possible to not bundle libraries. In the [OpenSCAD package](https://codeberg.org/guix/guix/src/commit/23f787dad6b48ad8d4b175c775258a6355d8a323/gnu/packages/engineering.scm#L3345) for example, all submodules are deleted and replaced with separate packages. I know, it's convenient if people could just do `git clone --recursive` and be done with, but Guix tries to keep package sizes small and there are a lot of software projects where submodules are not regularly updated to prevent security issues.

Thus I would recommend not to add yet another submodule to OpenSCAD. At least as long as it's relevant at compile time.

2. For my Python projects, both at work and privately, I started using [pre-commit](https://pre-commit.com/) last year. The way this works is that you add a `.pre-commit-config.yaml` file to the source tree. After cloning the source the first thing you should do is run `pre-commit install`, which will install the hook in git.

What pre-commit does depends on what you configure. There are linters available for many languages (Markdown, Python, Rust, etc.) or more general hooks which ensure there are license headers on each source file or things like that.

Some hooks, upon commit, modify files (e.g. trim trailing whitespaces), in which case the commit is blocked, the changes are kept on the files and you can manually review and stage them.

It's pretty robust.

What it doesn't do however is preventing people from submitting code which violates style guides for example. as one could always not enable the pre-commit hook.

So in addition to that, there have to be github actions as well, which perform the same checks. What pre-commit does though, is tell you early so you're not only noticing when you create a PR for you 50 commit new feature that half of them have some style issues.

3. I wasn't able to find an uncrustify extension for pre-commit. I briefly considered writing one (after all, all it takes is a small YAML file), but uncrustify is written in C++, which is not in the list of supported languages for pre-commit extensions.

An option might be to switch to clang-format, which seems to be pretty widespread. Even though it also seems to be written in C, there is a pre-commit extension for it (so after all it does seem to be possible to write a pre-commit hook for uncrustify, but I'd rather not have to maintain yet another special tool).

There already is a `.clang-format` in the repo, which works with clang-format >19.

Running it on `src/**/*.(c|h|cc|cxx|hxx)` with the existing config produces quite some changes:

```diffstatt
528 files changed, 44183 insertions(+), 43539 deletions(-)
```

Running uncrustify produces a far smaller diff though:

```diffstat
196 files changed, 3115 insertions(+), 3150 deletions(-)
```

AaronVerDow, 2025-08-10:
I've gotten the workflow to create a patch artifact when running through GitHub or act. This way, no matter what development environment people are using, there will be a reliable way to get and apply the required changes.

t-paul, 2025-08-10:
> I plan to contribute any improvements back to the OpenSCAD project, once they are complete and stable enough.

I hope that at least considers the build flows OpenSCAD uses (e.g. older distros, build on OBS having no network access at build time, Docker based builds, ...), otherwise this will cause a huge amount of headache and/or collisions.

t-paul, 2025-08-10:
> I'm not getting much benefit out of modifications, so I'll be starting with unmodified LLVM as the closest style.

Meaning close to the style OpenSCAD uses (= similar to the uncrustify config)? Sounds like a good plan.

I don't know much about the other styles and in the end I would not care too much if it's not a totally crazy style like trying to fit C++ into some sort of K+R style.

The only suggestion I'd make in case there's a column limit somewhere in the style, then pushing that to something like 200 or so would make sense to me. I'm not sure how many people still work on 80 column VT100 nowadays but optimizing for that and accepting those horrible unreadable multi-line expressions seems to be a bad choice 🙊.

kintel, 2025-08-10:
I'm also fine with a liberal column limit for now. The only reasonable case for 80 cols I've seen is to be able to fit a 3-way diff on a single screen side by side. ..which happens quite a bit for the VSCode crowd.

t-paul, 2025-08-10:
> fit a 3-way diff on a single screen

Yeah, that one is probably too small on any normal monitor format always.

In general I do break long expressions into separate variables anyway to solve the issue of too long lines. But if there's one, the auto format is often just so horrible that a long line seems better when just reading the code. I have not looked at how clang-format does for this specifically, but I've seen the mess Eclipse did wrapping lines at literally the worst possible places.

@AaronVerDow the comparison table above, did that happen with or without ignoring white spaces? I don't know how common the current uncrustify setting of 2 spaces / no tabs is. Just so we don't miss an otherwise very similar style just because it defaults to 4 spaces or using tabs.

AaronVerDow, 2025-08-10:
+1 on longer columns, and I'll rerun the comparison without whitespace changes.

jordanbrown0, 2025-08-10:
I like to have **some** line length limit, so that I can set my window to some width and have it always be enough.

nomike, 2025-08-11:
> I hope that at least considers the build flows OpenSCAD uses (e.g. older distros, build on OBS having no network access at build time, Docker based builds, ...), otherwise this will cause a huge amount of headache and/or collisions.

I've created a package for PythonSCAD for GNU Guix and also took over maintenance of the OpenSCAD package. Guix is a Linux package manager/distribution which is fully bootstrapped has 100% reproducible builds and all builds happen in containers with no network access. Libraries are un-bundled wherever possible.

So yes, I will make absolutely sure that there is nothing which requires network access or other external dependencies.

What I'm currently doing in the package is to do a `git pull --recursive` and then throw away the whole "submodules" folder, just to ensure that there are no bundled libraries.
I would have left out the `--recursive` flag for git, but there is `libraries/MCAD` which is needed for building AFAIR.

AaronVerDow, 2025-08-11:
I created two pull requests, one just has the required changes, and the second one includes all of the formatting updates.

The points relevant to this discussion:
* Git hooks are supported but cannot be applied by default. A method for installing a hook is documented.
* There was an existing clang-format configuration file in the codebase that I had overwritten and forgot about. I am leaving that as is for now unless there are other suggestions.

kintel, 2025-08-13:
Fixed by #6084 and #6085

closed 2025-08-13 by kintel