openscad/openscad #6059: Redefine $fs and $fa as hints

jordanbrown0, 2025-08-01:
The relationship between $fs, $fa, and $fn is currently fully documented. I suggest that we respin that documentation to define $fs and $fa as hints that OpenSCAD is free to use, or not.

First, for non-circle cases (text, imports) the behavior is already unspecified. We try to do something sensible with the values to influence the resolution of the result, but we don't say what. We should do that with $fs and $fa too, promising only that the result will be approximately the same as the current definition, or better. 

In particular, it should be made clear that if in the future we have a geometry engine that can directly process circles, we would be free to ignore $fs and $fa (but not $fn) and generate true circles (and spheres and ...) instead of polygons.

We should continue to describe the current algorithm, so that people have a clue what good values might be, but we should make it clear that we might produce better results than that algorithm requires.

Again, $fn is different. It is legal and appropriate to use $fn to generate regular polygons, and it should always be respected.

Discuss.

jordanbrown0, 2025-08-01:
Similarly, but probably better a separate issue: we should explicitly disclaim any specific algorithm for generating spheres; we should be free to use a "better" algorithm (in particular, octahedral subdivision).

nophead, 2025-08-01:
I don't see why they should suddenly be described as hints when they have been obeyed since OpenSCAD was born.

I rely on them being obeyed so that my 3D prints have an appropriate number of facets to have the right detail but not swamp the serial link of the printer with thousands of tiny segments. Also my previews have the correct trade of between smooth curves and speed.

bitbasher, 2025-08-01:
may i extend the scope of change to these points:

1. change curved shapes to generate an optically correct and smooth surfaces, circles, text, etc. all the time
2. retain the math specification of a primitive along with its mesh so it can be regenerated 
3. always generate the mesh of object(s) considering intersections, diffs and unions with others 
4. add a regular polygon function - reg_poly( numsides=3 ) - to be able to stop using a side-effect of circle/sphere with a silly value for $fn to make one.
5. remove $fn entirely
6. allow the user to set properties on the model that give a "hint" about its intended use
Jordan and i have already established that we can agree vehemently on some topics of discussion, but in this case i completely agree and would extend the his suggestions to improve the user experience.

Speaking to the last item first, #6 is about giving the app "hints" about the intended use of the model. 
 hint 1 - status of the model design process - draft, final
 hint 2 - output quality - image - image size , format (VGA .. 4K .. 8k etc)
 hint 3 - output quality - 3D print - nozzle size, filament size, layer thickness, or any other setting to be passed to the printer (for file formats that support that sort of thing .. STL does not of course

My point #1 is to completely hide $fa-$fs, or even make them obsolete, by always producing a mesh that, from the viewpoint (or camera if you prefer), appears to be smooth. And when the viewpoint is changed, redo the mesh if the size of segments/polys has changed by more than a threshold constant.

We would need an algorithm that would consider:
- distance to the viewpoint
- size and curvature of the surface
- hints from the user about the purpose of the model (point #5)
- etc
to calculate how many line segments / quad polygons to use to render to preview or render an image / mesh file

from: [how small a thing can be seen](https://www.sciencefocus.com/the-human-body/how-small-can-the-naked-eye-see)

> humans can resolve two lines about 0.01 degrees apart: a 0.026mm gap, 15cm from your face. In practice, objects 0.04mm wide (the width of a fine human hair) are just distinguishable by good eyes, objects 0.02mm wide are not.

so a possible algorithm to use is:
`ob_angle = calculate_angle_subtended( object );
seg_angle = ob_angle/0.01; // the visual size of object / 0.01 - smallest thing the eye can see
seg_length = consider_curvature( object.size, seg_angle );
`
divide angle subtended in the current view by the whole object by the angle subtended by the smallest thing a human eye can see to get the number of angular segments, then convert that to a linear length to use as the length of line segments or polygon edges. Poss use the "draft" rendering hint to increase the Smallest Subtended angle to lengthen the segments and reduce their number.

#2 is to make #3 possible

#3 instead of leaving it to the script writer to call render() when needed .. build the mesh as objects are instantiated taking note of the scope of the operators they are inside of. In the scope of a difference operator, thus inside its braces {}, calculate the intersections and build the mesh with all of the edges of intersection included.
Likewise for intersection() and union(), but not for color().
Basically take note of the operations that apply to a shape to figure out how to make its mesh

#s 4 & 5 i think we are ready for reg_poly() and reg_prism() methods, and then we do not need $fn at all

#6 let the user give hints about the intended use of the model, quality metrics, etc so that the "decisions" made by the app are better guided by the intent of the script writer

jordanbrown0, 2025-08-01:
To be clear, I'm not suggesting any implementation change in the foreseeable future.

I want to clarify what is and is not guaranteed behavior.

The intent of these features is to control how OpenSCAD approximates circles.  (Again, $fn excepted.) If OpenSCAD can in the future approximate circles better, for similar performance on then-current systems, it should be free to do so.

Nobody should really care how many segments a *circle* is broken up into. A circle should be a circle, and anything less than that is an unfortunate implementation limit that might be improved in the future. 

I regard this as being along the same lines as the question of how and whether to reduce everything to triangles.

We should at the same time and for the same reasons explicitly make the default values be subject to change.

nophead, 2025-08-01:
Triangles v quads make no difference to the 3D printing results but changing circles will break everything. OpenSCAD is a tool for producing meshes, so there will never be true circles. The best you can so is put millions of segments into STL files but that is stupid. I make them huge and slicers would have to remove some to be 3D printable or convert them into gcode arcs.

jordanbrown0, 2025-08-01:
> changing circles will break everything.

An interesting perspective. I'm pretty sure that it would break none of my models, because if I want a circle, then I want a circle, and the higher the quality the better. If I want a regular polygon then I use $fn, which must not change. 

It seems to me that if your model's correctness depends on $fa/$fs forcing lower resolution circles, you're just doing it wrong. I have a hard time coming up with any case where it would be a good idea. 

Indeed, for practical reasons it is sometimes sensible to limit the number of sides - but those practical reasons are subject to change over time as technologies improve.  File and memory sizes that were unthinkable a few years ago are routine now, and similarly processing complexity that was once unthinkable is now routine.

> OpenSCAD is a tool for producing meshes,

No.  OpenSCAD is a tool for producing 3D models.

> so there will never be true circles.

Purely a limitation of current file formats. Current 2D formats *can* represent true curves. Future 3D formats might well be able to too.

Today's 2D drawing programs certainly act like they can do Boolean operations on curves.  Assuming that's what they are doing, there is an excellent argument that OpenSCAD's 2D subsystem should do the same, deferring breaking curves up into line segments until it's necessary to move into 3D. Doing so would, for instance, allow an SVG export to export native curves.

nophead, 2025-08-01:
It isn't a file format limitation. Both CGAL and manifold produce meshes as CSG results. And since polyhedron is a primitive meshes can be input and so must be output.

pca006132, 2025-08-03:
@nophead by "changing circles will break everything", what do you mean by "break" in this case? Not having the exact mesh, it will functionally break, or something? Do you have an example in which increasing the `$fs` or `$fa` will break things?

nophead, 2025-08-03:
I set $fa to 6 because 60 facets is enough for them to be not visible when printed in ABS. I sometimes lower it for shiny PLA filaments. It also has to make the number of facets divisible by 4 so that corners created by hulls of circles have the edges in the correct place.

I set $fs to extrusion_width / 2 so that I don't get segments that are so short they can't be represented by a filament but swamp the serial bus of my printers. For example, I printed a jigsaw piece that I scanned to an SVG and that had segments too short until I rounded it a bit with offset.

pca006132, 2025-08-03:
> I set $fa to 6 because 60 facets is enough for them to be not visible when printed in ABS. I sometimes lower it for shiny PLA filaments. It also has to make the number of facets divisible by 4 so [...]

It sounds that if things are divisible by 4 it should be fine? Making segments finer will not break the model --- it just doesn't work with the current slicer setting you use. I think modern slicers can set a minimal path length or something to simplify stuff.

pca006132, 2025-08-03:
Though, it sounds weird that you require the number of facets to be divisible by 4 for things to work. Does that mean you depend on where the segment starts?

jordanbrown0, 2025-08-03:
The fact that current 3D  file formats cannot handle true curves is most certainly a file format limitation.  Yes, any hypothetical future format would need to handle straight-line meshes, but that doesn't mean that it couldn't also handle true 2D and 3D curves, just as current 2D file formats can handle both curves and polygons.

Managing the connection to the printer and the printing resolution (in all axes, including linear along the perimeter) seems like it is properly a slicer function, not a function of a tool whose job it is to build a mathematical model.

The multiple-of-4 thing is a good, albeit minor, example of what I mean by treating these values as hints.  Today, we have strictly defined the math used with $fs and $fa.  (Which is, by the way, wrong - it measures along the arc when it is defined to measure along the chord.)  If we redefine them to be hints, with the exact math not a committed interface, we could change it so that the $fa/$fs calculation always rounded up to a multiple of 4.

---

The long-term goal should be that an unadorned circle(10) should produce a perfect 10-unit circle, without requiring any additional user input.  Currently, it produces (and is defined to produce) a 30-gon, which is not nearly enough to be perceived as a true circle.

I was thinking in terms of changing the math associated with $fa and $fs, but perhaps it would be enough to change the defaults.  Perhaps we could (soon) make the defaults smaller - perhaps $fa=2 and $fs=1 - and perhaps we could eventually make them tiny.  Or perhaps we could define `undef` to be whatever we expect to be the best circle practical - ideally, a true circle - and set the default to that.  Fiddling with the defaults would maintain compatibility for existing programs that set the values, while only affecting those programs that *don't* set the values.  Those few programs that actually want $fa=12 and $fs=2 would need to explicitly set them.

---

Any change here would likely want to be coordinated with #5570, adding a function that would make it unnecessary to do your own $fa/$fs/$fn calculations.

nophead, 2025-08-03:
If it ever changes I will have to redefine circle to put the old behaviour back. I can't see how you can ever represent true circles with the current back ends though.

nophead, 2025-08-03:
Yes the geometry engines. And having a perfect 2D circle that gets linear extruded into a polygonal prism would make no sense. Whether you like or not OpenSCAD is a mesh modeller.

jordanbrown0, 2025-08-03:
>Whether you like or not OpenSCAD is a mesh modeller.

Except for those people who use it to build models for their laser cutters, and those who use it solely to produce images of 3D models.

---

I agree that extruding a perfect 2D circle into a polygonal prism is questionable.  However, I'm not sure that it's any more questionable than the fact that the 3DP pipeline never even comes closing to building spheres; instead it builds stacks of finite-thickness cylinders.

It's also pretty questionable that when the user says "circle" they get something that is very clearly *not* a circle.  Small circles are particularly bad; pentagons are not circles no matter how hard you squint.

I think we should aim for the point where circle() yields as perfect a circle as we reasonably think it can, and extruding that circle yields as perfect a cylinder as we reasonably think it can, and I'm probably OK with those not being quite the same shape.

nophead, 2025-08-03:
Linear_extrude has to sweep exactly the 2D shape, or it doesn't make sense. It doesn't just work on circles but arbitrary polygons. And if a circle is not a polygon then you would get very strange effects. For example, if you union a perfect circle with a random polygon, how is that going to be represented and extruded. You can't even draw it with a graphics card unless it is made from triangles. 

I think you are dreaming if you think OpenSCAD can ever represent perfect circles. It has to produce meshes for downstream tools like slicers. And it has to produce meshes to be able to draw on the screen. So the best you can do is set $fn to some crazy high number, which you do already if you want. No need to change anything.

rcolyer, 2025-08-03:
> The relationship between $fs, $fa, and $fn is currently fully documented. I suggest that we respin that documentation to define $fs and $fa as hints that OpenSCAD is free to use, or not.

Wrongly.  It is currently documented wrongly, and as far as I remember has always been, despite repeated attempts to make it correct.  I remember at one point trying to make it factually correct, but people quickly found that too confusing.  At issue is trying to make it simple for non-programmers to understand, which has repeatedly failed.  e.g. it currently says, "$fa is the minimum angle for a fragment, the angle A-B-C in the diagram, and controls the division of large circles."  Unfortunately, this is false.
```
circle(40, $fa=61, $fs=1);
```
This produces a circle with 6 sides, and angles of 60 degrees, less than the 61 degrees that the documentation describes as the "minimum angle".  This is because it has never been the minimum angle, but everyone keeps trying to simplify it this way.  The same is true of $fs, where:
```
circle(40, $fa=1, $fs=20);
```
produces segments that are about 19 units long, while the documentation describes, "$fs is the minimum size of a fragment" for an $fs of 20.  What the actual behavior does, and has always done, is to give us circles "at least this good on resolution", but with a switchover between $fa and $fs giving the minimum number of segments specified by either the $fa or $fs constraint.  So this has always been an "at least this good" specification, we've just struggled to describe it that way in the documentation because of the confusion that arises with the interplay between $fa and $fs.  I think your suggestion, Jordan, is an opportunity to clarify what we mean by "at least this good".

I concur that both the intent and behavior of $fa and $fs has always been specifying a spatial and angular resolution/smoothness need, and then for optimization of runtimes letting OpenSCAD go as polygonal as it can as long as it meets at least the specified requirements for resolution.  The only thing that has ever differed from this is the factually incorrect phrasing in the documentation.  $fn is the variable used to specify an exact need, and has always been.

> I think you are dreaming if you think OpenSCAD can ever represent perfect circles. It has to produce meshes for downstream tools like slicers. And it has to produce meshes to be able to draw on the [...]

It might be a dream, but a plausibly achievable one.  We are rolling out a new engine right now, and in doing so have gotten ourselves a backend selector and some infrastructure for this.  It has also always been the case that the vast majority of the OpenSCAD language itself is not mesh-like and is almost entirely unconcerned with meshes.  The core of OpenSCAD as an abstract language has the right elements to support mathematically perfect abstractions with transformations performed on them.  There are also other output formats which exist (and others that could be created) to capitalize upon this abstraction, making many elements much more naturally smoothed in a highly space-efficient and computationally efficient manner, leaving it up to final output systems like slicers or CNCs to decide what to do with these elements.  We actually have a segment of potential users that routinely show up and then drift off to other solutions precisely because we only support meshes as outputs, they needed more than that, and they expected otherwise by the nature of the language.  It ends up not meeting this other set of needs.  So Jordan is correct that there is an opportunity for interested developers to expand the capability of OpenSCAD into this other valuable space, translating the abstract geometrical elements straight through into abstract geometrical elements or specified smooth curves or surfaces in the output.

jordanbrown0, 2025-08-06:
> there is no model being built is there .. the script is processed directly from module call to mesh

No. It goes from script to a CSG tree (there's a menu item to see the CSG, but I'm on my phone), and from the CSG to preview or mesh (or maybe other representations, not sure).

But there are very distinct "run the script" and "process the geometry" stages.

CSG as text looks like OpenSCAD text, but that's a reconstruction from the internal data structure.

jordanbrown0, 2025-08-06:
> So this has always been an "at least this good" specification

Eh, I'd say "about this good". It's pretty clear that if you say $fa=30 it shouldn't be producing a 60-gon, and if you say $fs=5 it shouldn't be producing 1-unit edges. I agree that the prose and the mathematical description disagree, but not by much. 

I'm starting to think that we should change the definition so that we can do things like round up to a multiple of 4, but for things like high(er) resolution circles by default, or true circles, we should be changing the default values, and in particular allowing for the possibility that the default might be `undef`. 

BTW, it looks like OBJ files can represent true curves. 

Note also that the whole discussion of printing practicality is assuming 1:1 scaling. Scaling up a small circle will produce stupidly low resolution, and scaling down a large circle will produce stupidly high resolution. A scheme that (somehow) defers polygonizing the circle to the last possible moment (preferably in the gcode arc processor in the printer) will be able to handle these cases much better.

coryrc, 2025-08-11:
> add a regular polygon function - reg_poly( numsides=3 ) - to be able to stop using a side-effect of circle/sphere with a silly value for $fn to make one.

Already done: https://github.com/BelfrySCAD/BOSL2/wiki/shapes2d.scad#functionmodule-regular_ngon

jordanbrown0, 2025-09-07:
>It's pretty clear that if you say $fa=30 it shouldn't be producing a 60-gon, and if you say $fs=5 it shouldn't be producing 1-unit edges. 

Actually, in the long run, if we can be confident that the performance wouldn't suffer, maybe it *should* generate higher-resolution circles than you asked for, treating it (as you said) as a "at least this good" specification.  If we could wave a magic wand and get true circles with no performance hit, it should probably just ignore $fa and $fs (but not $fn) and make true circles.

nophead, 2025-09-07:
No because it would flood the serial port of my 3D printers. That is why I set $fs to half the extrusion width. Shorter than that wouldn't affect the print quality but it would increase the file size, slicing time and Comms bandwidth. I like OpenSCAD because I can specify where every vertex in the model is.

jordanbrown0, 2025-09-07:
Managing your printer's serial port is not OpenSCAD's job.  Get a better slicer.

If you want to specify where every vertex is, don't ask for a circle.  Circles don't have vertexes.

jordanbrown0, 2026-11-15:
#6890 adds fragments(), which is defined to tell you the number of sides that a circle with particular parameters will have.  This allows models to move away from duplicating the $fs/$fa calculation and to work with $fe, and allows changing the details of the calculation without affecting models that only want to match the sides of round things.

nophead, 2026-11-15:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

$fs and $fa are needed for 3D printing.

jordanbrown0, 2026-11-15:
Nobody ever said "get rid of $fa and $fs" - though in theory that should be the goal; circles should be circles, not polygons.  ($fn excepted; it's well-defined and useful as a way to create n-gons.)  But we are nowhere near there.

But any library that thinks that it knows what $fa and $fs mean will soon(?) be wrong, because if $fe is set it will take precedence.

fragments() allows a program to be independent of the exact calculation; it will take into account all current and future algorithm changes.

In addition, existing $fa/$fs calculations are awkward:  they will produce results with non-multiple-of-four vertexes and so will not have vertexes on all axes.  (For instance, they will not automatically align with octahedral spheres.)  Rounding up to the next multiple of four makes things fit together better without significantly changing the precision, but is not currently compatible because programs depend on the details of the $fa/$fs calculation to align the sides of various round things.  We shouldn't make this change now, because of that compatibility issue, but I think that we *should* make it immediately after the next release, giving applications an entire release cycle to adopt fragments().

Finally, the existing $fa/$fs defaults were set something like ten years ago, for a now-ancient generation of hardware and a far, far slower geometry engine.  They should be updated to more appropriate values for today; we should not doom all future users to crummy circles by default.