openscad/openscad #6066: Separate evaluator thread

pca006132, 2025-08-03:
**Is your feature request related to a problem? Please describe.**

Complicated scripts can take a long time to evaluate. The current evaluation is done in the GUI thread, which makes the GUI unresponsive.

**Describe the solution you'd like**

Run the evaluation in a separate thread, and provide ways to stop the evaluation.

closed 2025-08-03 by pca006132

gsohler, 2025-08-03:
> duplicate.

Not wanting to toss here.

Reference implementation for a separate Evaluatgor thread can be found in

https://github.com/pythonscad/pythonscad/blob/master/src/gui/CSGWorker.cc

and its links into

https://github.com/pythonscad/pythonscad/blob/master/src/gui/MainWindow.cc

its very much a clone of CGALWorker (name is not appropriate anymore)

pca006132, 2025-08-03:
@gsohler is it doing preview? do you think it will be easy to upstream it?