openscad/openscad #6074: Simplifying the generated CSG files

gtoal, 2025-08-05:
Mainly for fun, I hacked up a little program today to take a .csg file and remove most of the redundant blocks.  I'm not saying it's particularly useful but you might find it interesting.  Have a look at 'flatten.g' in http://gtoal.com/OpenSCAD/simplifycsg/

There are still features I want to add (merging cascaded transformation matrices being the biggest one); and it's not been extensively tested; it's more of a demonstration program to show what's possible, and I just thought you might like to see it.  There's an input file test.csg in that directory and the current output is in test-out.csg

My reason for writing this was to clean up some messy and mostly ad-hoc code in one particular openscad file I was working with, and although similar logic could potentially be added to openscad's csg writer, I realise there's not a lot of benefit from or justification for adding something like this to OpenSCAD, so this post is more of an FYI than a serious request to add the functionality to the package.

Graham

pca006132, 2025-08-05:
We do this automatically in the manifold backend, and it does more stuff that requires knowledge about the geometry.
For the CGAL backend, I think this is not done, but idk how many people care about its performance now.

gtoal, 2025-08-05:
> We do this automatically in the manifold backend, and it does more stuff that requires knowledge about the geometry. For the CGAL backend, I think this is not done, but idk how many people care about its performance now.

Oh cool, I didn't realise that the manifold code affected anything other than the graphical display!  I'll give it a try, thanks.

gtoal, 2025-08-05:
> Well, it does it internally, and will not affect the csg tree that openscad shows. It is only an optimization that makes things faster, i.e. doesn't change the csg file.

Ah.  Well I don't know if you've looked at the generated CSG much, but there is a huge amount of noise in there - 80% of the text is dead code - redundant groupings, empty groups, etc.  I guess it doesn't matter when it's only ever being looked at by other programs, but it kind of rankles against my sense of aesthetics when I look at it by eye :-)

jordanbrown0, 2025-08-06:
Mixed feelings. There is a ton of noise in there that could be automatically elided - I'm pretty sure that the one-child "group" nodes could easily be automatically hidden when viewing or exporting. On the other hand, it directly represents the internal data structure, and is tied to the program flow, and those are interesting things too.

pca006132, 2025-08-06:
> Ah. Well I don't know if you've looked at the generated CSG much, but there is a huge amount of noise in there - 80% of the text is dead code - redundant groupings, empty groups, etc. I guess it [...]

Make sense, I was thinking about it in terms of faster csg, not from readability.

kintel, 2025-08-10:
I keep manually cleaning up csg files, mostly for internal testing, so I'd say it's an interesting idea to add options to OpenSCAD's csg exporter to expose some options.

#1881 is also related