openscad/openscad #6075: Every link in the cheat sheet is now broken

amatulic, 2025-08-05:
**Describe the bug**
Recently an editor took it upon himself to massively rewrite and reorganize the OpenSCAD documentation wiki, breaking every single link in the OpenSCAD cheat sheet. It is not clear to me that this user knows anything about OpenSCAD either.

This is serious. The cheat sheet is my primary entry point into the documentation, as it is for other OpenSCAD users.

**To Reproduce**
1. Go to cheat sheet at https://openscad.org/cheatsheet/
2. Click on any link
3. Observe that no link leads to a valid target any longer.

**Expected behavior**
The links should work.
Either all the edits over the past couple of weeks need to be reverted (simplest solution) or the cheat sheet needs to be updated to use new anchors.
My preference is to revert. The new organization isn't necessarily better, and the new styling definitely isn't.
I left a note on the editor's talk page [here]{https://en.wikibooks.org/wiki/User_talk:VulcanWikiEdit#Please_STOP). If I don't get a response from that user in a couple of days, I will proceed with reverting the recent changes to the wikibook.

amatulic, 2025-08-05:
The edits have been going on for a couple of weeks. I planned to find the points on each page where the links started to break and revert back from there. Another option is to fix all the headings so the links work again, but I don't feel it's my job to incrementally clean up the mess; it's better to revert back.

amatulic, 2025-08-05:
I'm thinking an easier way would be to insert `{{anchor}}` tags back into the wikibook in appropriate places, and leave the cheat sheet alone. This would be easier than trying to figure out new anchors in the cheat sheet for everything. The person who broke the links could help out here. It shouldn't be hard; I estimate I could get the 2D and 3D primitive anchors in the cheat sheet fixed in less than half an hour, because many of the links in the cheat sheet point to the same document on the wiki.

amatulic, 2025-08-06:
@bitbasher Thanks for responding.

At the moment, I would like you to go to https://openscad.org/cheatsheet/ and start with the first link, fixing the target page for every single one. I have already fixed all the links in the "2D" and "3D" panels by inserting anchor tags on the target pages.

For example, "circle()" on the cheat sheet has the target https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem#circle -- but you removed the "circle" heading from that page, breaking the link. Therefore, just above the new heading "Circle Object Module" I added the tag `{{anchor}circle}}` to restore the `#circle` target from the cheat sheet.

Other cases weren't as simple. For example, "polyhedron()" on the cheat sheet points to https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron but you didn't just change the heading to break the link, you removed the entire section into a transclusion page. I added the tag `{{anchor|polyhedron}}` above the new heading "Polyhedron() Object Module" to the transcluded page and inserted back into the original target page. I had to do this in other cases too.

I also fixed the "Operators" panel; you didn't break that, but some headings had changed in the past and broke some links there.

In the "Syntax" panel on the cheat sheet, I couldn't figure out how to fix the "var" link. I also couldn't figure out how to fix anything in the "Special Variables" panel, at which point I was tempted to revert the entire target page to the original.

Therefore, I ask you to go through **all** the links and fix them ASAP, reverting yourself if necessary, giving priority to the functionality of the currently-live cheat sheet at the expense of your changes. The cheat sheet is the primary entry point into the documentation and it no longer works. I wish you had kept this in mind as you were making changes so that nothing broke in the cheat sheet.

bitbasher, 2025-08-06:
i can do that .. if i need to make any big reversions i will capture my work in to my sandbox and restore the links

bitbasher, 2025-08-07:
wow, thanks @amatulic for your help!!

bitbasher, 2025-08-08:
> Thanks. ... The documentation for `$preview` has been lost.
not lost .. i moved it to make a new page about Rendering in my sandbox.. i have pasted my updated text back into Other Lang Features

> The three links in the **Lists** panel of the cheat sheet are still broken
i made a new page for Vectors .. it is now transcluded into the General page so those work again

> The **Mathematical functions** panel has broken links because you removed some mathematical functions from the page
i had input from Jordan and others about a few things being wrongly lumped into math functions 
i rem let() in particular

 why? I notce the trigonometry functions are gone. Please restore them.
fixed

i have tested all the links now and i think all are working

bitbasher, 2025-08-08:
i believe i have now checked every link .. someone with permission to close this can do so.