openscad/openscad #6082: Documentation `each`, examples fail compilation

pkriens, 2025-08-10:
The `each` documentation does not compile. The each only works inside a list declaration but it is shown outside several times. I.e. `each [...]` is wrong, it is `[ each ...]`. ( I think)
```
echo( each [1:2:5] ); 
echo( each [6:-2:0] );
```

[image]

Clearly this needs fixing in the documentation asap but I wonder if we should make it legal?

UBaer21, 2025-08-10:
`[each [1:2:5]]`   the result need to be a list (range get converted).
1,2,5  is neither a list nor string or has any data format - so this could only be allowed for `echo()`

UBaer21, 2025-08-10:
fixed: https://en.wikibooks.org/w/index.php?title=OpenSCAD_User_Manual/List_Comprehensions&stable=0#Each_Element

jordanbrown0, 2025-08-10:
As for making "each" work outside list comprehensions, it could in theory be used as a mechanism for a spread syntax - converting a list of values into arguments to a function or module.  It would be OpenSCAD-ish there, but unlike any other language.  And "each <obj>" could convert an object into named arguments.  But there wouldn't be any way to create arguments where order was important between positional and named arguments, or where a particular name is duplicated... both valid for echo().

Not this issue, of course.