openscad/openscad #6091: resize() module docs do not explain its arguments

bitbasher, 2025-08-14:
**Describe the bug**
The documentation for the [resize module ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#Resize)is incomplete

**To Reproduce**
follow the link and read the section
there is no explanation of the variations on the auto argument, nor why convexity needs to be an argument when resize can only make the object bigger or smaller, and cannot change its convexity

**Expected behavior**
I expect to learn why  **auto** argument can be a number or a vector, and why convexity needs to be an argument at all.
I expect to be warned that for the directions i do not want the shape to change in zero needs to be given in the elements of the vector

jordanbrown0, 2025-08-14:
`auto` cannot be a number.  However, like many (or most, sigh) type-error cases, if you supply a number it is simply ignored.  If you supply a list of numbers, it looks like they are interpreted using usual truthy/falsey rules, so zero is equivalent to false and non-zero is equivalent to true.  That inconsistency is irksome.  I expect that the inconsistency is accidental, but am not sure whether the treatment of lists-of-non-booleans is intentional.

Saying there is no explanation is an overstatement.  There's several sentences on the subject.  However, they are indeed not clear.  What does "to match" really mean?  If you say `resize([7,3,0], auto=true)`, what does that really mean?

It does say (in the second sentence) that any axis with a zero value is left unchanged.

Looking at the source...

* `auto` is internally called autosize.
* `auto=true` is equivalent to `auto=[true,true,true]`.
* The default is [false,false,false].  (Which is equivalent to a scalar false.)
* The autosize flag for a particular axis is consulted only when the resize value for that axis is zero.
* When the resize value for an axis is zero and the autosize flag is true, the autoscale value is used.
* When the resize value for an axis is zero and the autosize flag is false, that axis remains unchanged.
* The autoscale value is the scale value determined for the largest new-size axis.  (That is, if scaling an object to [1,2,3] it is the scale for the Z axis, and if scaling to [3,2,1] it is the scale for the X axis.)

I have not really internalized what that behavior *means* and so can't supply a good prose description at the moment.

The `auto` value is a list so that you can have different behaviors for different axes.

WRT convexity (which isn't documented, but perhaps should be)... I assume (without inspection) that the answer is that resize() behaves like the other operations that fully render their children, and loses the underlying children's convexity.  The general rule about convexity is that any operation that can work without rendering, that doesn't have to know the actual mesh to do its job, doesn't need convexity specified, and any operation that requires the actual mesh (resize, minkowski, render, polyhedron, import-of-stl, et cetera) do need it. (Hull generates a mesh, but that mesh is always convex.)  In the particular case of resize, it probably could just inherit the children's convexity.  (Perhaps it does; that would require research that I don't have time for at the moment.)

jordanbrown0, 2025-08-14:
Historically, you wanted to avoid resize() if you could, because it was slow to preview.  scale() does not need to render its children, because it doesn't care how big they are; resize() needs to render them because it *does* care how big they are.  With the advent of Manifold that difference is far less important.