openscad/openscad #6094: Beta flatpak from 2025-08-14 shows a stoneage version / breaks BOSL2

mstefani, 2025-08-14:
**Describe the bug**
The OpenSCAD beta flatpak from today shows the version:
"OpenSCAD 2011.11.11 
https://www.openscad.org/
Copyright (C) 2009-2025 The OpenSCAD Developers"

That version produces an assert in BOSL2:
Compiling design (CSG Tree generation)...
[ERROR: Assertion '(version_num() >= 2.01905e+7)' failed: "BOSL2 requires OpenSCAD version 2019.05 or later." in file .local/share/OpenSCAD/libraries/BOSL2/std.scad, line 8](https://github.com/openscad/openscad/8,/home/michi/.local/share/OpenSCAD/libraries/BOSL2/std.scad)
[TRACE: called by 'assert' in file .local/share/OpenSCAD/libraries/BOSL2/std.scad, line 8](https://github.com/openscad/openscad/8,/home/michi/.local/share/OpenSCAD/libraries/BOSL2/std.scad)

**To Reproduce**
Steps to reproduce the behavior:
1. Update to the flathub-beta from 2025-08-14
    Commit: 4dc2b49d27bf966de808a6e32fcf58983611876a95b245c29b294b4f716ab122
    Parent: 73ccf3d2b114fab3ef48848d5c40aad6c8c36633579e31c9a78db5fdd8569865
    Subject: Merge pull request #54 from flathub/bump-versions (2b7b5b11d788)
    Date: 2025-08-14 15:51:40 +0000

3. Start OpenSCAD
4. Look at console for the version
    "OpenSCAD 2011.11.11"

**Expected behavior**
A version starting with "2025" or at least newer that the last stable "2021.01"

**Code reproducing the issue**
include <BOSL2/std.scad>
cyl(d=10, h=10);

**Environment and Version info (please complete the following information):**
 - OS: Fedora 42
 - System: x86_64
 - OpenSCAD Version: beta flatpak 4dc2b49d27bf966de808a6e32fcf58983611876a95b245c29b294b4f716ab122

**Library & Graphics card information**
BOSL2: git commit 0fdf90938e32f73b742fea0adb705b2028ead742

t-paul, 2025-08-14:
Yes, I have seen that. No idea why they are trolling like that, it looks like the build machines have that set as current date.

Also, wrong repo. I'll close this one, correct repo would be https://github.com/flathub/org.openscad.OpenSCAD but as we don't own that, I can't move the issue.

closed 2025-08-14 by t-paul

t-paul, 2025-08-14:
Fixed via https://github.com/flathub/org.openscad.OpenSCAD/pull/57 - it may take some time to propagate to the beta channel.