openscad/openscad #6111: macOS: OpenSCAD doesn't bring its window(s) into view when the Space they occupy is not in view.

Zenmaster43, 2025-08-19:
**Describe the bug**
I use macOS Sequoia with OpenSCAD 

**To Reproduce**
Steps to reproduce the behavior:
1. Configure multiple desktops in macOS Spaces.
2. Launch a different application in the one Space.
3. Use the command-tab keyboard combination to switch to another Space.
4. Launch OpenSCAD from the Finder.
5. Open or select a file or create a new one in OpenSCAD.
6. Use command-tab to select the other open application. (that application and its containing Space come into view and receive focus /become activated.)
7. Use command-tab to select OpenSCAD. (OpenSCAD becomes activated but  its containing Space remains out of view.
8. Manually select the Space containing the OpenSCAD window using command-tab and clicking on the OpenSCAD icon. ( the Space does switch here, since that's a feature of macOS Finder, independent of the apps)
9. The OpenSCAD window remains behind other windows if any were in front of it. (By clicking OpenSCAD icon, at least the frontmost OpenSCAD window should have been activated here.
10. Sliding three fingers "up" on the trackpad shows the desktop with miniature images of all windows open on the active Space.
11. Click on the image of the OpenSCAD window. (Then it finally activates.)

**Expected behavior**A clear and concise description of what you expected to happen.
7. Use command-tab to select OpenSCAD. (OpenSCAD and its containing Space should come into view and OpenSCAD and its frontmost window should become activated.) This removes the need for steps 8-11 above and brings the GUI into conformity MacOS User Interface Guidelines and general expectations of users.

**Code reproducing the issue**
(not related to user code)

**Screenshots**
(not practical due to dynamics of UI while using command-tab. It would require a video recording)

**Environment and Version info (please complete the following information):**
 - OS: macOS Sequoia 15.1
 - System: MacBook Air M2
 - OpenSCAD Version 2024.11.10 (git 6f5932feb)

**Library & Graphics card information**
Terminal doesn't run the application even when I enter OpenSCAD --info when in the /Applications/OpenSCAD.app/Contents/ directory. Using the Library Info menu item and resulting window only allows copying one line at a time, even though I selected all lines of the info window. (Too complicated to piece it all together that way.)

**Additional context**
Within the application's GUI, it all works nicely, but with relation to the Finder, it doesn't cooperate with most of the non-basic features of switching apps and Spaces. It only really response to a launch or a click in its windows to reactivate, requiring the user to manually select the Space and then manually click the OpenSCAD window. Also, the Window menu in OpenSCAD doesn't provide a list of open windows to select from, which is a pretty standard feature for most macOS apps for several years already. This makes totally no sense at all to me.
 Copy should capture all of the selected text in the info window, rather than just the first line of the selection.

kintel, 2025-08-19:
Could you re-test this with a more recent snapshot? Using 2025.08.14 on macOS 14 (Sonoma), the space switching works as it should. Haven't tested on Sequoia yet..

Note: To show info, run `/Applications/OpenSCAD.app/Contents/OpenSCAD --info` verbatim in a terminal.

kintel, 2026-09-03:
I believe this got fixes when we upgraded to Qt6. Could you check in a more recent build?