openscad/openscad #6114: OpenSCAD is creating non-manifold models

dan-p3rry, 2025-08-20:
**Describe the bug**
OpenSCAD dumps a polyhedron with non-manifold edges.  The non-manifold behavior is present with both output processors (manifold/CGAL).  OpenSCAD shows no warnings or errors during render.

**To Reproduce**
Run this scad file, then import STL to prusaslicer or meshlab.  (In meshlab, Filters-->Selection-->Select non-manifold edges.

**Expected behavior**
clean output

**Code reproducing the issue**
```
include <BOSL2/std.scad>

$fs = $preview ? 1 : 0.4; 
$fa = $preview ? 3 : 1;

manifold_check();

module manifold_check() {
    path = subdivide_path(rect(13.5, chamfer=1), 25);
    path_sweep(mask2d_chamfer(x=1, spin=180), reverse(path3d(path)), 
            closed=true);
}

```

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11 up-to-date
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version 2025.08.18

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

UBaer21, 2025-08-20:
This is an BOSL issue not open SCAD.
The polyhedron is created by BOSL.
https://github.com/BelfrySCAD/BOSL2/

UBaer21, 2025-08-20:
https://github.com/BelfrySCAD/BOSL2/issues/1777

dan-p3rry, 2025-08-20:
If it's a BOSL2 polyhedron issue, why is OpenSCAD not complaining about non-manifold design?  Why does it render a non-manifold polyhedron without error message?

nophead, 2025-08-20:
Yes, unless you do a CSG operation on it. Try unioning it with a unit cube. You should get an error from CGAL.

dan-p3rry, 2025-08-20:
I tried .3mf, still non-manifold according to prusaslicer.  (Meshlab cannot import 3mf, at least it doesn't recognize the file suffix).

kintel, 2025-08-20:
My bet is that PrusaSlicer has a too strict definition of "manifold". Perhaps they convert internally to STL or smth.?

kintel, 2025-08-20:
Faces in the wrong direction sounds unlikely to me, so some digging is needed to prove that's the case. I'd assume self-intersection is the culprit, but it's really hard to evaluate PrusaSlicer's manifold check from a distance.

Perhaps a good starting point would be to slim down this model to the minimal example, e.g. no chamfers, nice coordinate values, minimum face count, then manually go through each face to verify.

adrianVmariano, 2025-08-20:
This looks like a BOSL2 bug to me.

dan-p3rry, 2025-08-20:
I don't know if this helps, but too much info is better than too little....  The actual endcap portion of path_sweep is difference'd out of the final model.  The bad faces after the difference() correspond to two of the three perpendicular faces in the last screenshot. 

And a question:  the suggestion that the issue is due to coincident endcaps is reasonable.  But people create coincident faces in their models all day long.  So the issue is that the polyhedron is coincident to itself?  

Thanks to everyone putting in effort on this issue.

dan-p3rry, 2025-08-21:
> And I am curious: It seems that the manifold backend thinks the mesh is manifold, i.e. no error? If you export the mesh as 3mf or obj, will prusaslicer and meshlab still complain?

I tried the export to 3mf.  PrusaSlicer still complained.  My version of meshlab doesn't support 3mf, but I saw a note that it was updated last month.  I'll update and try and report back.

dan-p3rry, 2025-08-21:
> Quick Q: How do you get PrusaSlicer to check for manifoldness?

Import the STL, then check the right pane.  If there's any non-manifold errors, you'll see something in the places I highlighted

[image]

kintel, 2025-08-22:
One interesting observation: I cannot reproduce the PrusaSlicer issue on macOS, neither with STL or 3MF..