openscad/openscad #6141: Display live cursor coordinates in 3D view

ftlpilot, 2025-08-29:
It would be extremely useful to display the current cursor coordinates, many editors/CAD programs do this in the bottom status bar, perhaps openSCAD could too.

kintel, 2025-08-29:
I assume the request was about displaying the text editor cursor position, not any 3D coordinates, making this a pure GUI task.

ftlpilot, 2025-08-29:
No – It would be useful (an adjunct option to distance measuring)

To have the cursor position also give the X/Y/Z CO-ORDS of the screen 3D model, perhaps only when it’s over a 3D surface, similar to the way Distance works

 

This email has been checked for viruses by Avast antivirus software.
www.avast.com

jordanbrown0, 2026-05-08:
I rendered and started a measurement, as I hover around I don't see coordinates either.