openscad/openscad #6148: MacOS render not complete from 2025-08-18 snapshot (missing some holes)

revk, 2025-08-31:
**Describe the bug**

The attached file works on snapshot 2025-08-17 and not on 2025-08-18. There are holes in the design (difference) which no longer work. An # prefix does show where the holes should be.

**To Reproduce**

Just render this file in each snapshot. Should have holes.

**Expected behavior**

Correct render

**Code reproducing the issue**

[Recorder.zip](https://github.com/user-attachments/files/22065391/Recorder.zip)

**Screenshots**

Good (2025-08-17)

[image]

Bad (2025-08-18) - note much faster

[image]

**Environment and Version info (please complete the following information):**

MacOS

[image]

Snapshots 2025-08-17 and 2025-08-18

**Library & Graphics card information**

**Additional context**

I appreciate it is snapshots - I hope I have tracked it down enough for you.

Thanks.

revk, 2025-08-31:
Can you clarify how I do that please.

kintel, 2025-08-31:
[image]

revk, 2025-08-31:
I'll test and let you know (tomorrow)

revk, 2025-09-01:
Sorry for the delay, yes, that is indeed the difference. Latest daily (2025-08-30) with CGAL works correctly. I hope that helps.

jordanbrown0, 2025-09-01:
Yeah, I noticed the automatic generation and lots of minkowski(). :-)

Here's my current minimal test case - generates a CGAL error in Manifold mode(!) but not in CGAL mode.
```
minkowski()
{
    union() {
        translate([1,0,0]) cylinder(d=2,h=1,$fn=16);
        translate([3,0,0]) cylinder(d=2,h=1,$fn=16);
    }
    sphere(r=0.25,$fn=6);
}
```
I believe the problem is that the two cylinders are touching along one line, which is usually a no-no.  Strangely, ordinary CGAL isn't complaining, but the bit of CGAL that's still in the Manifold minkowski() processing *is* complaining.

A workaround is, in module m0(), in the "if(hole)" case, change the diameter of the cylinder to 2.0001.

jordanbrown0, 2025-09-01:
You could probably move them a micrometer without practically affecting alignment.

Yeah, I haven't tracked down exactly why the problem with the small round holes also affected the large rectangular hole, but I think they are both part of larger operations.  (They are probably part of the same difference, and maybe even part of the same minkowski().)

The trick to isolating things like this is to find *one* error shown in your model, and chop away stuff until that one error goes away, then put back the last thing you removed, and try removing something else.  Repeat, repeat, repeat.

You say there's another hole that's not fixed by tweaking the size of that cylinder? Where?

While it's good that we've got you a workaround, we still have a problem that, at least pending somebody else identifying it as a known problem, needs to stay open.

revk, 2025-09-01:
Sorry it is fixed now with the tweak to size. My issue is that it is not only the holes with the clashes that failed, all holes failed. The fix to just the LEDs fixes all holes.

I also mean I cannot "move" as the location comes from the PCB, and that is where the LEDs are, but changing size has fixed, thanks.

If I can help in any other way, please say.

revk, 2025-09-01:
Glad you found a simple test case, and I hope that helps.

jordanbrown0, 2025-09-01:
Yes.  The odd thing is that it fails in Manifold mode but succeeds in CGAL mode.

pca006132, 2025-09-02:
Will have a look at this, but as @kintel said, if it is related to CGAL import, we can't do much about it (other than trying to implement Minkowski/convex decomposition in manifold).

pca006132, 2025-09-03:
It is stuck inside the CGAL convex decomposition part. I have no idea how this can be a valid Nef polyhedron, considering it has self-intersection (the shared edge). I guess this makes CGAL's algorithm confused? Seems like a bug in CGAL's convex decomposition code.

pca006132, 2025-09-03:
iirc CGAL will reject shared edges/vertices as self-intersection. This is what happened in other failed Minkowski, when we need to convert the manifold mesh to CGAL Nef polyhedron for convex decomposition. For this case, the conversion succeeded, but the convex decomposition is extremely slow for an input with only <60 faces: it took at least 1 minute, and I did not wait long enough to see if it would finish. I am pretty sure the mesh in manifold's output is duplicated appropriately, as it satisfies the topological requirements for being a 2-manifold.

kintel, 2025-09-15:
Apart from Jordan's hang, this is a duplicate of #5866. Let's discuss the problem there, and once solved, we can get back to the original design posted here.

closed 2025-09-15 by kintel