openscad/openscad #6152: Visual Studio Build Setup Issues

kintel, 2025-09-01:
This is a collection of various issues/topics related to getting a Visual Studio build going by following https://github.com/openscad/openscad/blob/master/doc/win-build.md

It's meant as a dump of topics which hopefully can be solved relatively easily. Larger topics should be split out into separate tickets.

* [ ] [Dependencies] In general, can we install dependencies from a script or package manager too, to avoid having to click-install, add to PATH etc. for each tool? ..plus allow us to manage dependency versions? This includes git, cmake, flex/bison
* [ ] [Dependencies] Git installation instructions are unclear. The link kind of recommends GitHub Desktop, but that doesn’t seem to install cmd-line git, at least it’s not clear how to make it work. Perhaps a specific PATH needs to be set? Installing the alternative git from git-scm yields an installer with a number of options. Perhaps the default options are good, but we should probably make that explicit
* [ ] [Dependencies] CMake is required to use vcpkg - we should add that as a dependency.
* [ ] [vcpkg] Our doc says to add vcpkg to PATH, but vcpkg docs also says to set VCPKG_ROOT. Should we do that?
* [ ] [build script] `win-msvc-build.bat` doesn’t break on first error, but keeps going - makes it hard to spot issues
* [ ] [build script] Packages get installed into the openscad source folder; we should move this somewhere more global, so that it doesn’t get wiped accidentally and so that multiple clones can use the same cache.
* [ ] [Packages] Boost: Unnecessary to install all boost libs; just install the ones we need
* [ ] [Build] Fails with `Could NOT find FLEX (missing: FLEX_EXECUTABLE)`, despite it being in the PATH

Sparsh-N, 2025-09-02:
I'll look into the other issues once I'm free this weekend, but looked at the readme again and it mentioned

```
If you're using a newer version of Ubuntu, you can install these libraries from aptitude. If you're using Mac, or an older Linux/BSD, there are build scripts that download and compile the libraries from source. Follow the instructions for the platform you're compiling on below.

A C++ compiler supporting C++17
[cmake (3.5 ->)](https://cmake.org/)
[Qt (5.12 ->)](https://qt.io/)
[QScintilla2 (2.9 ->)](https://riverbankcomputing.com/software/qscintilla/)
[CGAL (5.4 ->)](https://www.cgal.org/)
[GMP (5.x)](https://gmplib.org/)
[MPFR (3.x)](https://www.mpfr.org/)
[boost (1.70 ->)](https://www.boost.org/)
[OpenCSG (1.4.2 ->)](http://www.opencsg.org/)
[GLEW (1.5.4 ->)](http://glew.sourceforge.net/)
[Eigen (3.x)](https://eigen.tuxfamily.org/)
[glib2 (2.x)](https://developer.gnome.org/glib/)
[fontconfig (2.10 -> )](https://fontconfig.org/)
[freetype2 (2.4 -> )](https://freetype.org/)
[harfbuzz (0.9.19 -> )](https://www.freedesktop.org/wiki/Software/HarfBuzz/)
[libzip (0.10.1 -> )](https://libzip.org/)
[Bison (2.4 -> )](https://www.gnu.org/software/bison/)
[Flex (2.5.35 -> )](http://flex.sourceforge.net/)
[pkg-config (0.26 -> )](https://www.freedesktop.org/wiki/Software/pkg-config/)
[double-conversion (2.0.1 -> )](https://github.com/google/double-conversion/)
```

I was wondering if we should also update some of the dependency minimums. Some of the one's I've noted just from a cursory look at the parent CMakeLists.txt file is 
- Double Conversion has no minimum mentioned (would this be an issue, considering most package installers provide a recent enough version?)
- Likewise, no minimum version enforced for PkgConfig, Flex, Bison, Libzip, Harfbuzz (MSVC Specific). That said, it may also not be needed if most package managers provide a recent version and some of these variables are printed to output for help when troubleshooting.
- `FontConfig` package in cmakelists is checking 2.8.0 minimum, but above notes 2.10. Could update?

Its tough to keep documentation in sync with the requirements, if there was a way to set these across the project it would be helpful.

Sparsh-N, 2025-09-07:
Took a look at some of the issues this weekend

I'm trying to broadly categorize each issue, but will look into fixing the non vcpkg dependencies first. I think the main vcpkg changes you've noted are good/easily mergable.

## Dependencies / Tools

- [ ] [General]: Tools seem to offer various options for shell in Windows (MSDOS, PowerShell, bash). Not sure what is actually common/preferred among devs - should we standardize on one of these? Same question goes for which terminal to use. Is Command Prompt the way to go?

For this part, I would say Powershell works fine for running the `win-msvc-build.bat` file, although in terms of personal preference, I find using bash nicer. When I was running or testing build steps, I would be using the integrated terminal in VS Code (ie no need to open Terminal externally), and I believe Visual Studio has the same support. It allows me to select the terminal to launch from there, and usually when developing the build steps, I would just use Powershell.

That being said, I did find myself occasionally having to open a bash terminal for installing certain packages. I agree that specifying which terminal to use in the instructions would be useful, although if we do switch to using bash, I could change the .bat script to .sh. I'm not sure if something has to be switched/customized to installing vcpkg.

- [ ] [Dependencies] In general, can we install dependencies from a script or package manager too, to avoid having to click-install, add to PATH etc. for each tool? ..plus allow us to manage dependency versions? This includes git, cmake, flex/bison

I could add a seperate build deps script for windows that installs these tools, although for flex/bison I remember installing from source. Mostly following the instructions here - https://github.com/lexxmark/winflexbison.

- [ ] [Dependencies] Git installation instructions are unclear. The link kind of recommends GitHub Desktop, but that doesn’t seem to install cmd-line git, at least it’s not clear how to make it work. Perhaps a specific PATH needs to be set? Installing the alternative git from git-scm yields an installer with a number of options. Perhaps the default options are good, but we should probably make that explicit

I agree, I'll revise the instructions. I suppose this, along with the other notes regarding documentation could be pushed in a single PR. (Can add cmake installation instructions here as well)

## vcpkg

I think these changes could be set in one PR. To my understanding, the main question I have is should I add/require setting VCPKG_ROOT and adding to PATH permanently (refer to description below)

- [ ] [vcpkg] Our doc says to add vcpkg to PATH, but vcpkg docs also says to set VCPKG_ROOT. Should we do that?

I think we should do both, although the tutorial provided here only provides instructions to set it for the current terminal session. I could note that `[System.Environment]::SetEnvironmentVariable("VCPKG_ROOT", "Path\to\vcpkg", "User")` would set it beyond the existing session, but not sure if the instructions should account for this. Most of the time if I add something to PATH I would manually go to the GUI for `Edit or Add System Environment Variables`.

The official doc mentioned - https://learn.microsoft.com/en-us/vcpkg/get_started/get-started?pivots=shell-powershell

- [ ] [build script] win-msvc-build.bat hardcodes d:/vcpkg as the install location. Use VCPKG_ROOT instead?

This is a easy change and I agree. It wouldn't work if VCPKG_ROOT variable is unset though, so it would be important to require users add/manually set VCPKG_ROOT permanently beforehand.

## build script

- [ ] [build script] win-msvc-build.bat does both dependency install, configure and make. This isn't really a typical dev workflow, rather a CI workflow. We should document each step instead like we do on other platforms.

I'm a bit confused, would this mean adding further comments in the script as to what each part does. I think we could remove the dependency install since it is using vcpkg in manifest mode, so when running the first cmake call (configure), it runs vcpkg install internally, referencing the `vcpkg.json` which is contained in the project.

- [ ] [build script] Packages get installed into the openscad source folder; we should move this somewhere more global, so that it doesn’t get wiped accidentally and so that multiple clones can use the same cache.

Just to clarify, this is referring to vcpkg_installed folder that is set in /openscad/? Sometimes it also installs within the build folder.

- [ ] Packages] Boost: Unnecessary to install all boost libs; just install the ones we need

I agree with this fix, I'll have to look into building/specifying the specific boost libs needed. I looked through the documentation and cant see too much, and looking at the CMakeLists I can't see the specified parts. Removing the libraries not needed would speed up the whole process, so I'll look into this more by next week.