openscad/openscad #6160: 3MF with lazy union fails to export and cashes

owebeeone, 2025-09-03:
**Describe the bug**
Loaded an Anchorscad generated scad file with a lazy union and attempted to export to 3mf and openscad just closed leaving a zero length 3mf file.

When running from the command line I get:
```
Fontconfig error: Cannot load default config file: No such file: (null)
Geometries in cache: 39
Geometry cache size in bytes: 669384
CGAL Polyhedrons in cache: 13
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.077
Top level object is a list of objects:
   Objects:    2
EXPORT-ERROR: Can't add triangle to 3MF model.
```

The command executed. (SCAD file attached)
```
['C:\\Program Files\\OpenSCAD (Nightly)\\openscad.exe', '-o', 'generated\\output\\anchorscad_models\\basic\\pipe\\anchorcad_Pie_default_example.stl', '-o', 'generated\\output\\anchorscad_models\\basic\\pipe\\anchorcad_Pie_default_example.3mf', '--backend', 'Manifold', '--enable', 'lazy-union', '-o', 'generated\\output\\anchorscad_models\\basic\\pipe\\anchorcad_Pie_default_example.png', '--autocenter', '--view', 'axes', '--imgsize', '1280,1024', 'generated\\output\\anchorscad_models\\basic\\pipe\\anchorcad_Pie_default_example.scad']
Exit code: 3221226356
```

**To Reproduce**
Steps to reproduce the behavior:
1. Load scan file
2. render
3. Export to 3mf
4. Crash

Or run the command line given above,

**Expected behavior**
Render the file (like it was working for a nightly build a few months older).

**Code reproducing the issue**
attached file is pretty short.

**Screenshots**
none to give

**Environment and Version info (please complete the following information):**
 - OS: Windows 11 Home
 - System: AMD 64
 - OpenSCAD version 2025.09.02 (git a16fcd05e)

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.
OpenSCAD Version: 2025.09.02 (git a16fcd05e)
System information: Microsoft Windows 11 (10.0.26100) x86_64 16 CPUs 31.26 GB RAM
User Agent: OpenSCAD/2025.09.02 (git a16fcd05e) (Microsoft Windows 11 (10.0.26100) x86_64)
Compiler: GCC "14.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_85
Eigen version: 3.4.0
CGAL version, kernels: 6.0.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.5.2
Manifold version: 3.2.1
Qt version: 5.15.17
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.85.3
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.16.0
freetype version: 2.13.3
harfbuzz version: 11.4.1
cairo version: 1.18.4
lib3mf version: 1.8.1
[... 420 lines cut ...]
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

**Additional context**
Was working on a previous build.

owebeeone, 2025-09-04:
Oh, this seems to be isolated to rotate extrusions. A bunch (not all) of rotate extrusions are failing.

kintel, 2025-09-04:
lazy union is not actively maintained, so issues are expected to creep in there. At some point we'll likely scrap that feature, or rather extract the most important part and bring it into OpenSCAD proper.

Anyway, if you find anything that's suspicious also with lazy unions turned off, please open a separate issue!

owebeeone, 2025-09-04:
I just ran the script with lazy union turned off and tried it from the GUI as well and it fails in both cases (lazy union turned off). All of the failures are rotate_extrude. I have a way of turning rotate_extrude into a polyhedron, as a polyhedron openscad exports to 3mf properly so I suspect it's an issue with rotate_extrude breaking something.

Running from command line...
```
Geometries in cache: 9
Geometry cache size in bytes: 173424
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.009
Top level object is a 3D object (PolySet):
   Convex:       no
   Triangles:    576
EXPORT-ERROR: Can't add triangle to 3MF model.

Command failed: ['C:\\Program Files\\OpenSCAD (Nightly)\\openscad.exe', '-o', 'generated\\output\\anchorscad_models\\basic\\spherical_cap\\anchorcad_SphericalCap_default_default_physical_example.stl', '-o', 'generated\\output\\anchorscad_models\\basic\\spherical_cap\\anchorcad_SphericalCap_default_default_physical_example.3mf', '--backend', 'Manifold', '-o', 'generated\\output\\anchorscad_models\\basic\\spherical_cap\\anchorcad_SphericalCap_default_default_physical_example.png', '--autocenter', '--view', 'axes', '--imgsize', '1280,1024', 'generated\\output\\anchorscad_models\\basic\\spherical_cap\\anchorcad_SphericalCap_default_default_physical_example.scad']
Exit code: 3221226356
```

```
// Start: lazy_union
default_5_default_5();
// End: lazy_union

// Modules.

// 'PartMaterial undef-default - default 5.0'
module default_5_default_5() {
  // 'None : _combine_solids_and_holes'
  union() {
    // 'default : _combine_solids_and_holes'
    union() {
      // 'cage'
      %multmatrix(m=[[1.0, 0.0, 0.0, 0.0], [-0.0, -1.0, -0.0, -0.0], [-0.0, -0.0, -1.0, -0.0], [0.0, 0.0, 0.0, 1.0]]) {
        // 'cage'
        color(c=[0.0, 1.0, 0.35, 0.4]) {
          // 'cage : _combine_solids_and_holes'
          union() {
            // 'cage'
            sphere(r=30.0, $fn=16);
          }
        }
      }
      // 'cap'
      multmatrix(m=[[0.0, -1.0, 0.0, 0.0], [1.0, 0.0, -6.123233995736766e-17, 0.0], [6.123233995736766e-17, 0.0, 1.0, 0.0], [0.0, 0.0, 0.0, 1.0]]) {
        // 'cap : _combine_solids_and_holes'
        union() {
          // 'cap'
          rotate_extrude(angle=360.0, convexity=10, $fn=16) {
            polygon(points=[[0.0, 30.0], [3.103306385869395, 29.839059795432938], [6.173316320767047, 29.35796596502815], [9.177090602896605, 28.56188033141773], [12.082400695762118, 27.459344373583413], [14.858074519318485, 26.06218758236497], [17.474330906034066, 24.385400537748456], [19.903099133370333, 22.446974069732278], [22.118320104303724, 20.26770622846981], [24.09622594441935, 17.870979134773002], [25.81559501568237, 15.28250810522525], [27.257979610755765, 12.53006574362732], [28.40790388485317, 9.643183959094847], [29.253029901447253, 6.652837107958736], [29.78429001027165, 3.5911096591488763], [29.995984137286854, 0.4908519487946045], [29.885840942752367, -2.6146722824297437], [0.0, -2.6146722824297473]]);
          }
        }
      }
    }
  }
} // end module default_5_default_5

```

kintel, 2025-09-04:
Thanks, this might be a duplicate of #6088 - could you check if the minimal example there also fails?