openscad/openscad #6161: SVG export test results differ across CPUs

kintel, 2025-09-04:
SVG export yields different numerical results across CPUs. This causes tests, e.g. `ctest -R export-svg_spec-paths-arcs01`, to fail when run on a different CPU than where golden files were generated.

This is quite visible on macOS, where CI runners run on Intel, while most dev machines are running Apple Silicon.

The tests in question were introduced in #6057, which added a number of SVG export tests and uses actual SVG files as goldens (comparing them as text comparison) rather than PNGs, as some of the tests are testing SVG export options which are not importable back in OpenSCAD.

Suggesions for fix:
* Carefully construct SVG option tests using simpler primitives which are less likely to yield different SVG results.
* Reduce number of SVG text comparison tests to a minimum and use export-import-to-png as a strategy for managing regression tests related to SVG export.

@AaronVerDow

kintel, 2025-09-04:
This is the failing test for `export-svg_spec-paths-arcs01`:
[image]

Interestingly, there are only 3 failing tests, all using the same file:
```
	2458 - export-svg_spec-paths-arcs01 (Failed)
	2480 - export-svg-fill-stroke_spec-paths-arcs01 (Failed)
	2502 - export-svg-fill-only_spec-paths-arcs01 (Failed)
```

AaronVerDow, 2025-09-05:
I compared in vimdiff with lines split and whitespace trimmed. The issue is that two new lines have been added. I'm assuming these additional lines are part of an arc, which would change the rest of the lines in that arc. If so, this could be a valid representation of the original drawing, but I want to poke around a bit more to see if I can find a reason why the number of segments would differ between the two drawings.

```
vimdiff <( cat ./tests/regression/export-svg/spec-paths-arcs01-expected.svg | sed 's/L/\nL/g' | sed 's/\s*$//' | grep -v '^$' ) <( cat fail.svg | sed 's/L/\nL/g' | sed 's/\s*$//' | grep -v '^$' )
```

[image]

AaronVerDow, 2025-09-06:
I think the difference here is from the SVG import and confirmed the differences are arcs being converted to lines:

[image]

Working on PNG comparison.

AaronVerDow, 2025-09-13:
I can post a quick and dirty update for png comparison if needed, but if this isn't a pressing issue I would like to tidy up the tests a bit before adding complexity.

kintel, 2025-09-13:
It's not super pressing, so let's defer as long as you're actively pursuing it.

coryrc, 2025-11-07:
I'll offer another option: generate goldens using a known-good binary before comparing with the latest build.

whosawhatsis, 2026-08-24:
I went looking for the cause of this and think I have it: **floating-point contraction (FMA) on arm64.**

`path::arc_to()` in `src/libsvg/path.cc` is built from `a*b + c` expressions:

```cpp
double x1_ = cos_rad * dx + sin_rad * dy;
double cx  = cos_rad * cx_ - sin_rad * cy_ + (x1 + x2) / 2.0;
```

arm64 has FMA in its baseline ISA, so Clang fuses each of those into a single `fma` — one rounding
step instead of two, giving a slightly different result. The x86-64 baseline has no FMA instruction
available, so the same expressions round after every multiply.

That difference propagates into `delta`, which feeds `getCircularSegmentCount(rmax, delta)` (ending
in a `ceil()` in `src/core/CurveDiscretizer.cc`) and then:

```cpp
unsigned int steps = std::max(fn, static_cast<unsigned int>((std::fabs(delta) * 10.0 / 180) + 4));
```

A sub-ulp difference either side of an integer boundary flips `steps` by one — and since every
sample is `phi = theta + delta * a / steps`, one flipped integer re-samples the whole arc.

That is exactly what @AaronVerDow saw: "two new lines have been added... these additional lines are
part of an arc, which would change the rest of the lines in that arc." That reading was right, and
this is the mechanism underneath it. It also explains @kintel's finding that Intel vs. Apple Silicon
is the only variable that matters — it is the ISA, not the OS or the toolchain version.

**Confirmation:** rebuilding *only* the `svg` target with `-ffp-contract=off`, changing nothing else,
makes all four `spec-paths-arcs01` tests pass on an M1 Max.

**Reproduction across machines:**

| Where | Arch | Result |
| --- | --- | --- |
| macOS CI (`macos-15-intel`) | x86_64 | passes |
| Debian 12, local | x86_64 | passes |
| macOS M1 Max, local | arm64 | fails |

**Magnitude**, for `export-svg_spec-paths-arcs01`:

```
vertex count  expected=170  actual=172  (+2)
max distance from an expected vertex to the nearest actual vertex: 0.437210 units
bounding box  identical on both axes to 4 dp
```

Worth being precise, since it cuts both ways: it is not last-digit noise (vertices move up to 0.44
units on a ~93 × ~33 unit model), but it is also not wrong geometry — both polylines are valid
discretizations of the same ideal arc, just sampled at different parameter values. The defect is
reproducibility rather than correctness: identical input, different output bytes depending on the
CPU.

One incidental note on why this stays invisible to CI — `macos-tests.yml` excludes the arm64 runner:

```yaml
        exclude:
          # macos-latest runs on arm64, which has a broken SW renderer
          - os: macos-latest
```

That exclusion is for an unrelated reason, but its effect is that the one configuration where this
reproduces never runs in CI, while being the configuration most contributors are sitting in front of.

I've opened #6974 with the one-line CMake fix. Full `ctest` on arm64 goes from 3 failed / 2725 to **100% passed out of 2758** with it. To be clear about how it relates to the direction
already discussed here: the PNG-comparison work @AaronVerDow started is independently worthwhile,
and if you'd rather go that way, this analysis is the useful part and the PR can be dropped. The
distinction is just that the test-strategy changes make the test stop failing, while this makes the
output stop depending on the CPU.

I have **not** checked whether the same class of divergence affects core tessellation —
`CurveDiscretizer` is used by all curved primitives and lives in a different target. There is no
failing test there, which isn't the same as being architecture-independent.

closed 2026-09-29 by kintel