openscad/openscad #6163: Logging of File load operation is missing

MichaelAtOz, 2025-09-04:
**Describe the bug**
Nightly is not logging File loads to console

**To Reproduce**

Steps to reproduce the behavior:

Start New OpenSCAD (recent version)
  File/Open - select a scad file
or
  File/Recent-files - select a file
or
  File/Examples - select an example
or
  File Reload

In Nightly, nothing shows in Console.

**Expected behavior**
In 2021.01 those four operations log to the console e.g.

> OpenSCAD 2021.01
> https://www.openscad.org/
>
> Copyright (C) 2009-2021 The OpenSCAD Developers
> This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
> 
> Loaded design 'C:/Users/me/Documents/OpenSCAD/ML-code/Hello-world.scad'.
> Loaded design 'C:/Program Files Other/OpenSCAD-2025.07.31-x86-64/examples/Basics/logo.scad'.
> Loaded design 'C:/Program Files Other/openscad-2021.01/examples/Basics/CSG.scad'.
> Loaded design 'C:/Program Files Other/openscad-2021.01/examples/Basics/CSG.scad'.

**Screenshots**
In this shot both windows had a File/Open operation:

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - OpenSCAD Version 2021.01 compared to 2025.07.31 & 08.31 from openscad.org

**Library & Graphics card information**
N/A?

**Additional context**
Note: In [Mailing-list entry](https://lists.openscad.org/empathy/thread/OMJCBAJYAYBF4LSJQAD3EVRBPPHLYREM?hash=MYJP7AD3TA4OKWHDF7CXPYTAEGUGCPE7#MYJP7AD3TA4OKWHDF7CXPYTAEGUGCPE7) (the reply text by Nathan down below) mentions for 2025.08.31 that:

> when I open a model, the console says
> 
> Loaded design …
> Used file cache size: 6 files

That specific mention was what triggered me to notice, so it logs sometimes or on some system???
I'll ask what Nathan is using.

bitbasher, 2025-10-02:
I am in Windows 11  --

in the welcome screen
when i select a file from Recent List or an example i do see the Loading message in the console

with the app open 
when i open a recent item, or an example i do see the loading message
and when i do File > Open i see the loading message

i have not seen a loading message when doing a Reload File .. but i do not change the file outside of scad and i imagine that would be needed to actually reload .. 

what is interesting is seeing the file load twice .. this is a small experiment i was doing .. when i open it from the splash screen's Recent List i see it two times in the console .. first without the loading msg, the second time with:

> Parsing design (AST generation)...
> Compiling design (CSG Tree generation)...
> ECHO: "003", "030", "300"
> ECHO: "example001 one two"
> Compiling design (CSG Products generation)...
> Geometries in cache: 0
> Geometry cache size in bytes: 0
> CGAL Polyhedrons in cache: 0
> CGAL cache size in bytes: 0
> Compiling design (CSG Products normalization)...
> Normalized tree has 1 elements!
> Compile and preview finished.
> Total rendering time: 0:00:00.420
> Loaded design 'D:/repositories/concat-recursive.scad'.
> Compiling design (CSG Tree generation)...
> ECHO: "003", "030", "300"
> ECHO: "example001 one two"
> Compiling design (CSG Products generation)...
> Geometries in cache: 0
> Geometry cache size in bytes: 0
> CGAL Polyhedrons in cache: 0
> CGAL cache size in bytes: 0
> Compiling design (CSG Products normalization)...
> Normalized tree has 1 elements!
> Compile and preview finished.
> Total rendering time: 0:00:00.221

i then opened the same file in another editor .. added a comment line and did File > Reload
i did not get a message .. in fact it updated the text in the editor but did not compile the revised script and gave no Loaded message .. so i dont think it reloaded anything .. i must still NOT be meeting the conditions for the reload to happen

bitbasher, 2025-10-02:
i was looking and another thing in MainWIndow.cc and came across the method that does the reload
```
void MainWindow::actionReload()
{
  if (checkEditorModified()) {
    fileChangedOnDisk();                  // force cached autoReloadId to update
    (void)tabManager->refreshDocument();  // ignore errors opening the file
  }
}
```
When i open a file, modify it, and then hit reload it correctly asks if i am okay with the reload wiping out my changes, so i let it
The text reverts to the original .. but there are NO messages at all output.

The heavy lifting is done in refreshDocument() - it actually read the file and compares it to the editor's content and if different pastes what it just read into the editor content and emits a log message .. but the log message does not show up

so there is a real bug in there somewhere

MichaelAtOz, 2026-01-29:
(CGAL results missing only noticed once out of at least 50 odd tests of the examples, file load message is lost frequently)

MichaelAtOz, 2026-02-08:
#6620 seems to have fixed the `Loaded` messages.

```
OpenSCAD 2026.02.07 
https://www.openscad.org/
 
Copyright (C) 2009-2026 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version. 
Loaded design 'C:/Users/me/Documents/OpenSCAD/My Things/tests/IssueDemonstration.scad'.
 
Parsing design (AST generation)...
 
Parsing design (AST generation)...
 
Parsing design (AST generation)...
Loaded design 'C:/Users/me/Documents/OpenSCAD/Builds/OpenSCAD-2026.02.07-x86-64/OpenSCAD-2026.02.07-x86-64/examples/Old/example024.scad'.
 
Parsing design (AST generation)...
Loaded design 'C:/Users/me/Documents/OpenSCAD/Builds/OpenSCAD-2026.01.25-x86-64/OpenSCAD-2026.01.25-x86-64/examples/Basics/projection.scad'.
 
Parsing design (AST generation)...
```