openscad/openscad #6165: Degenerated face with manifold back-end from simple geometry

tturpin, 2025-09-05:
I'm just starting to use the (not so) new manifold back-end, which seems much more robust and efficient than CGAL, but I found some errors in meshes generated from some models. Here is the simplest one, which I reduced as much as I could. It generates a mesh (3mf or stl) with a degenerated face (checked with FreeCad).

Here is the code:
```
module profile() rotate(45+2) {
  translate([0,-0.8-0.5]) square([15,0.8]);
  translate([4.6,-0.8-0.5]) rotate(-45) square([2,1]);
}

linear_extrude(40,center=true) profile();
mirror([0,0,1]) translate([0,0,-20]) linear_extrude(5) profile();

```

Tested with OpenSCAD-2025.09.02.ai27105-x86_64.AppImage on Ubuntu 24

tturpin, 2025-09-08:
Thanks for looking at this !

So, the issue is obtained by taking the union of two identical linear extrusions, with different heights.

The generated 3mf contains:

> 					<vertex x="11.180736" y="10.083708" z="-20.000000" />
> 					<vertex x="11.180736" y="10.083708" z="15.000000" />
> 					<vertex x="11.180736" y="10.083708" z="20.000000" />
and
> 					<triangle v1="15" v2="17" v3="16" pid="1" p1="1" />

where 15,16,17 are the indexes of the shown vertices, which are obviously (exactly) aligned in the z direction. In this particular case, I would be surprised if it was related to rounding, but maybe, on the x/y coordinates ?

My main concern is with other software, as you said (FreeCAD), but note that it also seems to make CGAL unhappy too if I try to use minkowski on the result. Extending the example as follows:

```
module degenerated() {
  linear_extrude(40,center=true) profile();
  mirror([0,0,1]) translate([0,0,-20]) linear_extrude(5) profile();
}

minkowski() {
  degenerated();
  cube(0.001);
}
```

I get:

> 
> Rendering Polygon Mesh using Manifold...
> WARNING: [manifold] Minkowski failed with error, falling back to Nef operation: CGAL ERROR: assertion violation!
> Expr: e_below != SHalfedge_handle()
> File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
> Line: 418
> 
> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
> Expr: e_below != SHalfedge_handle()
> File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
> Line: 418

The actual model on which I was initially stuck with FreeCad (and couldn't find a working repair solution) is much more complex, and has different errors (self-intersections), but I started with the simplest 'failing' model that I could find, as it's probably easier to investigate a simple union like this than try to simplify the complicated one hoping to preserve the issue.

By the way, the "complex" model took several minutes to render with CGAL (with several cgal assertion failures) and under 2 seconds with manifold (without errors other than mesh issues), so, I wished I heard of manifold before and I'm really looking forward to see major breakthrough fully operational !

tturpin, 2025-09-18:
This is somewhat disappointing. I think that ensuring some kind a soundness should be a key objective of any OpenSCAD rendering replacement work. By soundness, I mean a clear definition of what the computed geometry object represents, what invariants they satisfy, what can assumed when using the resulting STL (or other) files outside OpenSCAD.

I've used OpenSCAD for a while, and I had resigned myself to the occasional CGAL assertions failures (which often require time-consuming magic workaround guesswork and sometimes effectively prevents the user from creating the geometry they wants in OpenSCAD).

Six years ago, I proposed a [draft implementation](https://github.com/openscad/openscad/pull/3152) of some consistency checks, in order to at least distinguish "bad inputs" issue from "real" CGAL issues, but never got any feedback on whether this improvement could be valuable or not.

So I was thinking that I had to accept the state of things, if I wanted to use OpenSCAD.

I only recently discovered "manifold" recently. By the way, I think that it would deserve more publicity, given how much it can improve user experience. I'm not in the OpenSCAD mailing list and I didn't see manifold mentioned in the OpenSCAD website (not even in the news section). And the latest stable version: 2021.01 is obviously much older. I use development snapshots, but since manifold is not the default, and there is no configuration step, this major option can stay unnoticed by potential users. I instantly saw an improvement over CGAL, both in terms of performance, but also robustness: my ongoing design was almost impossible to continue with CGAL errors, and went smoothly with manifold. This is definitely much better.

However, I quickly found that the generated STL had "errors" (the definition of which depends on what software reads them). But at least it has errors that prevent OpenSCAD itself from using some of its own generated manifold geometries, until all operations including, e.g., minkowski, are done with manifold.

Fortunately, slicers (I use Cura) appear top be quite permissive w.r.t. "problematic" STL files. But in my case, I wanted to perform a FEM analysis with FreeCAD, and it failed at random steps. I'm still impressed by FreeCAD's ability to manipulate inconsistant geometries silently, happily convert them to a "shape", then to a mesh, after which gmsh will either fail, or succeed creating a mesh with no volume, or an inconsistant mesh whose analysis will fail with some cryptic "non-positive jacobian" error.

I spent a week looking for solutions. I couldn't change manifold's generation, so I tried various different softwares to try to "fix" my STLs. Some work for some models, but not other, etc. FreeCAD cannot address all issues. I had some success with MeshMixer, which unfortunately is not free, after the trial period. Finally, I found that Mehslab's "Isotropic Explicit Remeshing" algorithm worked the most consistently (until now). Followed by "Quadratic Edge Collapse Decimation" with "Preserve Normals" and "Preserve Topology" options enabled to recover planar surfaces. This is obviously less than ideal.

I understand that formally specifying supported geometries is not trivial, with corner cases of legitimate connected polyhedra not representable as a single manifold mesh, and numerical precision issues, but I wanted to stress how useful I think it would be to have something well-defined, which "works", and not just "works most of the time". I didn't read the theoretical work behind manifold, so I don't know how realistic it is to ask for a formal specification and guaranteed consistent output, but believe that the (improving) lack of consistency is one of OpenSCAD's strongest limitations.

kintel, 2025-09-18:
To add to @pca006132's comments, I believe OpenSCAD outputs meshes which comply with 3MF's definition of manifold meshes, as described under https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#41-meshes

The two things we do not support fully are:

_All meshes SHOULD have:_
* _Minimal self-intersections: Producers SHOULD NOT produce self-intersections or overlapping objects; consumers MUST accept self-intersecting meshes and overlapping objects, but treat "model" and "solidsupport" types as if filled with a Positive fill rule (see next section)._
* _Non-degeneracy: All triangles SHOULD have a non-zero area, as this will create numerical instabilities while calculating the triangle normal._

In addition, OpenSCAD still struggles with being unable to perform minkowski sums on certain types of meshes due to CGAL's import interfaces having no support to import some corner case manifold meshes. This is one if the last issues keeping us from declaring the Manifold backend fully supported.

..and to repeat @pca006132's plea: Avoid STL!

kintel, 2025-09-18:
> Six years ago, I proposed a [draft implementation](https://github.com/openscad/openscad/pull/3152) of some consistency checks, in order to at least distinguish "bad inputs" issue from "real" CGAL issues, but never got any feedback on whether this improvement could be valuable or not.

Your work is interesting. I was under the impression that this was WIP with "a serious bug", so we marked the PR as "Work in progress", but no further work has pushed after that. ..then Covid hit and we all got other things to think about. That said, what you proposed is not particularly trivial, but I'd welcome better sanity checks, both for the polyhedron primitive and for imported files.

tturpin, 2025-09-18:
> For things to be 2-manifold, you may need to duplicate vertices, but STL throws away these information

I didn't understand that the inability to represent duplicate vertices was actually a cause for STL loss of information and possibly subsequent errors (as duplicate vertices are usually considered errors in other software). Now it makes sense. I will keep this in mind, lesson learned ! I couldn't solve my geometry issues my just switching to 3MF, though.

> We have a formal definition of correctness, it is just not the one you want. You can have a look at https://github.com/elalish/manifold/wiki/Manifold-Library

Very enlightening reading. I understand that manifold was actually developed with soundness considerations seriously in mind:

> Closing the design loop with powerful analysis tools like FEA and CFD in an automated optimization could vastly improve the efficiency of design[...] However, the lack of reliability of the computational geometry foundations [...] means this level of round-trip automation is not currently feasible.
> 
> The goal of this library is to create a robust computational geometry foundation from which advanced use cases like the above can be realized.
> 

Thanks for the definitions. I would love to see all this precisely detailed in the OpenSCAD documentation, once manifold is released. The "epsilon-valid" mesh concept makes the definition really non trivial, though, and this property is not directly usable from other software, I guess. I'm pleased to know that self-overlaps removal is the next item on the road map, and that you are working on it.

kintel, 2025-09-19:
I do have a workaround in mind though: I think I can construct a CGAL Nef polyhedron for a non-manifold mesh by building a union of all the triangles making up said mesh. Just need time to write up the fallback and creation code..

kintel, 2025-09-19:
@dan-p3rry It feels like it's slowly time to actively discourage people to export to STL, and drive them towards better export formats (both OBJ and 3MF should be reasonable alternatives I think). The best help in this scenario would be to help make sure that OpenSCAD->3MF->`some app` works well for tricky designs, and open tickets if not.

kintel, 2025-09-19:
* _regression w.r.t. CGAL in current stable version:_ **Not important**, but CGAL vs. CGAL regressions are.
* _regression w.r.t. CGAL in current snapshot:_ **Yes**
* _Regression when opening from other software, Opening issue with other software, wether it's à regression or not:_ **Yes, but** it's often hard to figure out if it's an OpenSCAD issue or issue in the other software, so a discussion is often needed first.
   * _And which ones (slicers seem rather permissive, FreeCAD much less), and which workflow:_ **It doesn't matter**, there are so many workflows, and we don't recommend specific workflows. But if some workflows appear to be very common, that would probably be a priority.
* _Also, a validity check to decide if an output is OK or not would be useful. Is there any such thing already implemented or even automated__ ?
   * _One basic test would be to try to import it in Openscad and possibly atempt some CGAL operation_: **We already do this automatically** for a lot of tests. If you find a problem, we should either match it with a known issue or create a new one.
* _Should we open a ticket for each failed test case (probably many) or collect them all in one place ?_ **One for each test case**, then we can merged them if they turn out to be variants of the same issue.

The same caveat as mentioned earlier exists: STL is a "one way street"; you cannot expect to be able to import an STL just because you exported it, as connectivity information is lost when exporting. For that reason, perhaps leave STL out of any such tests.

Btw., we have a bunch of very minimal corner case tests in https://github.com/openscad/openscad/tree/master/tests/data/scad/3D/misc. One sanity check to make is whether you discover an example which effectively reduces down to one of those examples.

tturpin, 2025-09-29:
> The best help in this scenario would be to help make sure that OpenSCAD->3MF->`some app` works well for tricky designs, and open tickets if not.

I did some testing on a large number of designs. I did not find many bugs, but still I reported a few issues. I could not automate FreeCAD compliance checking, so I only  tested a few models with this workflow and did not find any case which would "work" with CGAL and not with manifold. However, most OpenSCAD generated geometries (from my designs, at least) have some errors from FreeCAD's viewpoint, so a regression is not very likely.

coryrc, 2025-10-15:
> [@dan-p3rry](https://github.com/dan-p3rry) It feels like it's slowly time to actively discourage people to export to STL, and drive them towards better export formats (both OBJ and 3MF should be reasonable alternatives I think).

Right now lib3mf is still considered optional (#6132); if the above is true, I think it should not. Should also F7 now export 3mf or OBJ? I could open an Issue on this.

kintel, 2025-10-15:
An issue on making 3MF a required feature would make sense, just to tease out what it would take to make that happen.

kintel, 2025-11-10:
The challenge is that we don't know what happens downstream. If we know that the user is targetting some downstream app which cannot handle non-manifold STLs, yeah, popup makes sense.
Also, STL isn't necessarily reimportable. ..and being reimportable isn't necessarily a sign that the STL is bad.

pca006132, 2025-11-10:
We can just say that if the stl is not reimportable, then it is not manifold and may cause issues. No need to predict what the consumer will do.

kintel, 2025-11-10:
Yeah, that's a good warning to display in the console. I'm just a bit hesitant to do it as a GUI popup.

jordanbrown0, 2025-11-10:
Do remember that "manifold" and "Manifold" are two different things.  Lower-cased it's a [geometric property](https://en.wikipedia.org/wiki/Manifold); upper-cased it's a [software library](https://github.com/elalish/manifold).

pca006132, 2025-11-10:
The issue with stl is not about the Manifold backend, it is a general issue. And for the minkowski issue, it is not related to stl as well, it is a general issue.

Also, regarding 3mf not being accepted: Does the software accepts obj?

coryrc, 2025-11-10:
> I'm in the "I'll keep with STLs" for-now camp, because I have some tools which do not accept 3MF.
> Like the old free Netfabb Studio Basic (which still works on W11)

AFAICT it does support OBJ.

> The issue with stl is not about the Manifold backend, it is a general issue.

True, but it appears far more likely to occur with the Manifold backend *and* STLs. My purpose here is to clear the blockers for the next release. This has been brought up as an issue and so some decision should be made; given the previous statement of "To be honest, I don't think this can be resolved any time soon" then hopefully we can find a different workaround (hence my suggestion of UX changes), but as the OP and you've said, it's not STL-only, so perhaps UX around STL export is not enough.

>  And for the minkowski issue, it is not related to stl as well, it is a general issue.

I think we have #6297 to handle that issue?

> Speaking of default export format, ... I would suggest:

Could we take the topic of UI changes for export to #6285? (I know it's currently closed, but either we'll reopen or we'll be referencing it in a new issue, so it's a good place to post right now). I'm hoping we can concentrate this issue on getting a decision on the title of this Issue.

Would #3152 solve most of the 3mf/OBJ-export degeneracy, leaving mostly STL-inherent problems which can only have a UX solution?

kintel, 2025-11-10:
#3152 is pretty dead; I'd say open a new ticket for that topic and fix it separately.

Edit: Oh just realized the authorship connection @tturpin :) if you want to revive that PR let's chat!

stefano2734, 2026-08-24:
Manifold 3.5.2 is now in dev version. 
Perhaps it is solved or partly improved.

jordanbrown0, 2026-08-24:
Retesting it and reporting whether it appears to be fixed would be a helpful way to contribute.

jordanbrown0, 2026-09-26:
I don't duplicate this with 2026.08.29 on Windows.

I ran the original example, and generated an STL and a 3MF.  I examined both of them, calculating the area of each triangle.  They were all non-zero (and not even close to zero; the minimum was around 0.5).

In the 3MF I do find the three points mentioned, but I do not find a single face that connects them.

The extended Minkowski-based test case succeeds.