openscad/openscad #6183: "Save a Copy" working identically to "Save As.."

Nikorasu, 2025-09-10:
I'm currently using OpenSCAD-2025.09.08 Windows version, although I noticed this issue ages ago. Just hadn't gotten around to mentioning it until now.. Today it almost caused some lost progress.

The _Save a Copy_ option should only create a new copy of the current file, with a new name, without subsequent Ctrl+S saves altering that new copy. It's supposed to work like a manual _'make a quick backup of the current state'_ option, which is then left untouched after that, allowing you to continue working within the previous save name.

However at the moment, it's functioning identically to "Save As", where the new name becomes the working-file. Once you use the "Save a Copy" option, the working filename on the current TAB, will then change, as soon as you make any further alterations to your file. That should not happen, the filename on the tab should remain the old name, and future Ctrl+S saves should only write to that original name, not the copy.

I hope I'm describing this in an understandable way..

bitbasher, 2025-09-20:
in the dev snapshot i do not see any problem the Save operations

creating a new file, then doing Save works like Save As ... by opening the File Dialog with proposed file name Untitled.scad
Doing Save once the file has been saved in the file system just updates that file

Doing "Save As ..." Proposes Untitled.scad for new files and the existing file name if it has previously been saved

Doing "Save A Copy ..." works just like "Save As .." - poses Untitled.scad for new files and the existing file name if it has previously been saved but if the file name is changed the editor tab name is NOT updated. 

This looks like it is all correct to me

UBaer21, 2025-09-20:
try "save A Copy"  and then "save" or "ctrl s"  - you would expect that File A
Save a copy "B"  is then saved as A but it is now B like on a "save as". So your active Filename changed but should not.

bitbasher, 2025-09-20:
> try "save A Copy" and then "save" or "ctrl s" - you would expect that File A Save a copy "B" is then saved as A but it is now B like on a "save as". So your active Filename changed but should not.

Now i see it .. the editor tab has remembered the Save A Copy filename .. i think the bug is in the operation of the Save action .. it is remembering the filename it got from the Save A Copy

so that is a bug in my opinion too

bitbasher, 2025-09-20:
/assign-me

bitbasher, 2025-09-20:
indeed .. the problem is in the save function .. all of the save routines, Save As and Save A Copy both call this version of Save() :
`bool TabManager::save(EditorInterface *edt, const QString& path)`
as their final step

but that means that these updates are done at the end of Save in all cases
```
if (saveOk) {
    LOG("Saved design '%1$s'.", path.toLocal8Bit().constData());
    edt->parameterWidget->saveFile(path);
    edt->setContentModified(false);
    edt->parameterWidget->setModified(false);
    par->updateRecentFiles(path);
    edt->filepath = path;
```
which does not change the the name of the editor tab after a Save A Copy but does record the file path from it an keeps it for use when doing ^S Save .. but it should not

I can either clone the Save code to make "aaveNoUpdate()" that Save A Copy can use
OR
i add a boolean param to the argument list of Save to disable that block of updates in Save, making it safe for Save a Copy to use

which option would be best? 
my vote goes for Clone the code to make saveNoUpdate()

t-paul, 2025-09-21:
Cloning whole functions should be last resort as that in most cases increases the maintenance burden.

I have not looked at the code in detail yet, but in general there's always also a 2 more options

1) splitting the function / taking out the part that is not always needed. Depending on the number of calling places it might be possible to only do
```
if (TabManager::save(...)) updateEditorFilepath();
```

2) passing in a callback that can handle additional actions
```
TabManager::save(..., [&](std::string& newFilename) { ... });
```

I suspect here the 2nd option is just a more complicated version of the boolean flag, so probably not a good choice. It still might be worth looking at the 1st option.

bitbasher, 2025-09-27:
there is a problem .. when a copy of a file has already been made my code detects it and asks about overwriting it .. but does not give the chance to change the filename .. i will fix this

Nikorasu, 2025-09-28:
> It seems to me that it should behave exactly like Save As does: pops a Save dialog, and if you click [Save] with a name that already exists, pops a "Do you want to replace it" dialog, and if you say "No" it leaves you with the Save dialog, where you can change the name.

That's not the issue.
The issue is what happens afterward, if you use Ctrl+S to save at some moment, after using Save A Copy. Once you do that, the Ctrl+S will start overwritting that copy, rather than the original save. Copy is supposed to create a backup file, under a different name, which does not become the **new working save name**.
Whereas "Save As" changes the working save name, so future saves overwrite that name.