openscad/openscad #6184: Inclusion should be guarded so that can't include a file multiple times.

Ma-XX-oN, 2025-09-10:
# **Is your feature request related to a problem? Please describe.**
When including an scad code file that implements dependent function/modules/variables and that includes other code that implements dependent functions/modules/variables, it's possible to cause a mutual inclusion resulting in a warning.

An overly simplistic model:
`test-included.scad`
```
echo("hello from included.");
include<test-includer.scad>
```

`test-includer.scad`
```
echo("hello from includer.");
include<test-included.scad>
```

If I execute `test-included.scad` I get:
```
WARNING: Can't open include file 'test-includer.scad'. 
Compiling design (CSG Tree generation)...
ECHO: "hello from included."
ECHO: "hello from includer."
ECHO: "hello from included."
```
Which seem to be an feeble attempt at breaking the mutual self inclusion.  

# **Describe the solution you'd like**
What should happen with mutually included files is that when attempting to include a file that is already included / is currently being included, it should just not do it and continue on.  No warning need be emitted (could have some sort of CL flag if it is deemed that this warning might be useful for debugging, but probably be better done using `echo` statements).  This would only occur when using `include` to include source.  Nothing is changed when using it in an expression context.  E.g. `a = concat([ include<foo.data> ], [ include<foo.data> ]);` will still work as expected.

This should be fine in terms of a functional language as even if it were to be included, it would result in redefinition of functions/modules/variables which would be illegal.

# **Describe alternatives you've considered**
I noticed that an include can be wrapped in an `if` statement to disable it's inclusion:
```
if (false) { include<foo.scad>; }  // foo.scad will not be included
```
I found that was interesting (`foo.scad` only contained an echo() statement), so I naïvely attempted to use it as an include guard.  Which of course won't work because if `foo.scad` declares anything like functions and modules, it would be invalid as it would be like declaring them inside of a block.  Assignments won't work either because their scope would be limited to the `if` block they were defined in.

# **Additional context**
No other context information needed.

Ma-XX-oN, 2025-09-10:
No, `use<>` doesn't evaluate assignments or other root level code which is not useful for this use case.  Something like `include_once<>` which would be a conditional textual inclusion would work and shouldn't be too difficult to implement with a simple `std::set<inode>` (or equivalent) to track file inclusion.

jordanbrown0, 2025-09-10:
I'm doubtful that we would add a third inclusion mechanism.  (Unless, perhaps, it also solved the namespace problem... and then it wouldn't allow top-level assignment either.)

Ma-XX-oN, 2025-09-10:
Why wouldn't it allow top-level assignment?  I don't understand this aversion to top level assignment.  It doesn't seem to be a necessary feature when you can disable execution using a top level variable and a ternary operator for expressions or an if statement for module calls to disable such code that isn't to be executed.

jordanbrown0, 2025-09-10:
Note that that comment was in the context of a hypothetical namespace-enforcing inclusion mechanism.  Such a namespace-enforcing inclusion mechanism might allow top-level assignment, but the top-level assignments would be inside the namespace.  It would not be possible, when using a namespace-enforcing inclusion mechanism, to set a value **outside** that namespace.

And why would that be?  Because the entire point of a namespace-enforcing mechanism is to make it so that the stuff contained can't conflict with the stuff outside.

Ma-XX-oN, 2025-09-10:
It sounds like you are disregarding the need to have libraries that depend on libraries, and sometimes the same libraries.  When that happens, the same file will have to be included (or whatever other name is used), so it would need to not reevaluate constants.  Unless this hypothetical namespace allows reimporting under a different name maybe, generating a template system which would allow passing special values to initialise the imported base values?  It sounds overly complex IMO.

I simply want a way to include a file without having to worry about having code be reevaluated or warnings being spit out saying that the constant has already been assigned.

Ma-XX-oN, 2025-09-10:
If another ``include<filename>` is not wanted, how about `include<filename, once>`?

bitbasher, 2025-09-19:
There was a reason that C uses pre-processor macros to protect inclusion .. like
```
#ifndef XYZ_H
#define XYZ_H
struct Foo {
    int member;
};
#endif /* XYZ_H*/
```
i always loved how this was both an elegant and an ugly solution all at the same time.
We can't do this in scad - no macros - and i do not advocate adding that

I am just saying that there is a parallel here to how scad does include <> .. it is a lexical
and textual substitution mechanism that owes nothing to language semantics but still
enables modular and structured programming paradigms

OR .. is there a way? humm maybe:
```
if( XYZ-lib ) {
XYZ-lib = true;
struct XYZ{
    int member;
    };
} // XYZ-lib
```

so in myscript.scad i can do
`include <XYZ-lib.scad>`
and will only be included once, right?

bitbasher, 2025-09-20:
> The parallel between C `#include` and OpenSCAD `include<>` is reasonably accurate, though C `#include` is always in units of lines, and OpenSCAD `include<>` can be used almost anywhere. But while C has preprocessor-supported `#if` and friends, OpenSCAD doesn't.

> * The variable is undefined on the first inclusion.  (Actually, it's undefined for all inclusions.)
> * The assignment happens inside the `if()` scope and so isn't visible to the second inclusion.
> * You can't define a module or a function inside an `if()`.  And if you could, it wouldn't be visible outside that `if()`.

well bugger .. clearly i have not yet wrapped my head around Functional Programming .. rereading my little example I see that I should have known it would not work .. 

complied / interpreted code cannot do the job of a pre-processor and vice verse

thanks Jordan