openscad/openscad #6199: Save as doesn't give all the workings of navigating like every other program seems to give you (Windows11).

Confuciusofessex, 2025-09-13:
**Describe the bug**
Save as doesn't give all the workings of navigating like every other program seems to give you (Windows11).
The same goes for save a copy.
There doesn't seem to be a way of setting a default save location either. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'file'
2. Click on 'save as....'
3. See error

**Expected behavior**
See second image.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: AMD Ryzen 9 9300x 12-core
 - OpenSCAD Version 2025.05.16(git c76900f9a)  Installed from ?

[image]

[image]

MichaelAtOz, 2025-09-14:
...and the extra line of stuff above 'Organize' & 'New folder', arrows and path field et, missing on the first image 'Save file' (OpenSCAD) v 'Save as' (.odp) image.
But, on my OpenSCAD 2025.08.31 / Windows 11, the arrows etc are shown, for 'Save' (a new file), 'Save as' & Save a Copy'.

Confuciusofessex, 2025-09-14:
Could someone show me what they get? Then I can see if I can give a better reply.

The top picture shows what I get with openscad and the bottom one comes from openoffice. The openoffice has the navigation bar at the very top. The openscad does not have this. Why?

If someone can confirm (with a picture) they get the navigation bar on windows 11 (I'm on Version 23H2 (OS build 22631.5909). Then I'll look into my installation. But I do get the navigation with other programs! I'm trying to nail this down to find out what is causing this difference.

The default save path was really a secondary observation on my part. 

Than You.

jordanbrown0, 2025-09-14:
2021.01 looks identical.

kintel, 2025-09-15:
Re. Qt6, see #4909