openscad/openscad #6202: incomplete documentation on the User Manual Paths page

bitbasher, 2025-09-14:
the first section of this page has numerous flaws - this list is incomplete but i have selected what i feel to be the priorities:
```
    ResourcesPath
        Posix:  // invalid - not an OS - normal users will not be able to know if their OS is POSIX complient .. or is only partly complient.
//     Linux: - missing
        Mac OS X: OpenSCAD.app/Contents/Resources
        Windows:// missing

    DocumentsPath
        Mac OS X: [NSDocumentDirectory], typically $HOME/Documents
// there is no NSDocDir defined in the shell environment, or anywhere else a normal user would find it
 .. and it should be
                $HOME/Documents/OpenSCAD

        Windows XP: [CSIDL_PERSONAL], typically C:users\\username\documents\GitHub\OpenScad\
// there is no CSIDL_PERSONAL defined in the shell environment
// how would a "GitHub" folder be part of a user's local documents?

// should remove as Vista is deprecated
        Windows Vista->: [CSIDL_PERSONAL], typically C:\users\username\documents\GitHub\OpenScad\

    UserConfigPath
// there no folder for scad in the User's app support, nor in the System library app support folders:
        Mac: [NSApplicationSupportDirectory], typically $HOME/Library/Application Support/OpenSCAD

// the D&S directory has not been used since before Windows 10
        Windows: [CSIDL_LOCAL_APPDATA], typically C:\Documents and Settings\username\Local Settings\Application Data
```
Sources for only _some_ of the color schemes are mentioned

References to the Windows Registry should carry a warning to not mess with it unless you really have to, better to just not mention it at all
----
I am working in Mac OS X version Sonoma 14.7.6 with scad snapshot 2025.07.11
 and  Windows 11 with scad snapshot 2025.08.18

bitbasher, 2025-09-14:
i use two of the scad platforms and have a copy of the page in my sandbox to update for my own purposes ... currently a big mess but when available it could be the basis for an update.

My first change is to remove all of the XDG references .. the freedesktop standard is far from widespread and the scad installation does not define any OpenSCAD* environment variables, much less the XDG* variables so i don't think it relevant to document them.

OR .. if the XDG env-vars _are_ uses in the code when available then that should be explicitly mentioned in the docs, and i will put them all back into my sandbox page

https://en.wikibooks.org/wiki/User:VulcanWikiEdit/sandbox/Paths

t-paul, 2025-09-15:
> My first change is to remove all of the XDG references .. 

Hu? What? Why?

> the freedesktop standard is far from widespread

I'm not sure that's even close to reality.

> if the XDG env-vars _are_ uses in the code

They are, which is why they are documented, e.g. https://github.com/openscad/openscad/blob/ee40fe50d23401e2f19b17ba5c0bcc9f13bacf54/src/platform/PlatformUtils-posix.cc#L35-L45

jordanbrown0, 2025-09-15:
For these purposes, "Posix" is a synonym for "UNIX-like".  That is indeed less jargon-y.  Changed.

The resources location is based off of the installation directory, and on Windows I think all you could give is the default install path.  Changed to "installation directory, typical default C:\Program Files\OpenSCAD".

I'm not really familiar with MacOS, but from context I suspect that there is no one answer, that the path to a user's documents is controlled by the system and only usually points to $HOME/Documents.  (And is it really $HOME from the environment, or is it the user's home directory?)

Windows is definitely like that:  the path to the user's documents directory is controlled by the system and is not visible as an environment variable.  The `CSIDL` names were obsoleted in Vista; the current name is `FOLDERID_Documents`.  That definitely *does* vary depending on the system configuration - in particular, on whether the user has used OneDrive to create a shared documents folder.  On Windows, probably the right answer is `[FOLDERID_Documents]\OpenSCAD`.  Changed, leaving a reference to CSIDL_PERSONAL since that's what's in the source.

Note:  I don't see any evidence that OpenSCAD uses the Documents path on Windows  for its intended purpose, the default location for documents.  I'd regard this as a bug.  It *does* use the Documents path for libraries and backups.

The references to XP and Vista should both be replaced by a single generic reference to Windows.  Done.

`CSIDL_LOCAL_APPDATA` has been replaced by `FOLDERID_LocalAppData`.  `C:\Documents and Settings` is long since obsolete; it should be documented as `[FOLDERID_LocalAppData]\Openscad`.  Typically it's `C:\users\username\AppData\Local\OpenSCAD`.  I'm not sure that that folder is created by default; I have one but I might have created it as part of experimenting with templates.

As for messing with the Registry... The danger is overblown.  Either that, or the danger of mucking with config files in UNIX-y systems is underblown.  The answer for either of them is that it's better to use the UI to manage them, and don't muck with them if you don't understand them.

The only color scheme that I see missing from the directory given is "Cornfield", the default.  That's not stored in any file; it's built in.

jordanbrown0, 2025-09-15:
The `OPENSCAD*` environment variables are documented so that people can set them if they choose.

jordanbrown0, 2025-09-15:
After some code inspection and experimentation, I find that indeed the UserConfigPath is not created by default.  So noted.

bitbasher, 2025-09-15:
oh .. and about the XDG environment variables .. my saying they are not in wide spread use was based on me setting up the XDG_* env vars to facilitate an installation of an app ported to Windows from Linux .. it really needed them, but in the process i found out that there is no OS that uses the freedesktop spec by default when setting up new users. Every platform lets a user define them in a shell start up script .. even Windows and Mac .. but they are only going to be set by power users .. that OpenSCAD uses them when they exist great .. but even the apps code has to check and have a backup path when they don't.

If these the XDG env vars are of real value for regular users then we should be providing detailed instructions on setting them up on all platforms .. OR at least giving links to other sources of instruction. Just putting them in a list of paths is enough for power users .. but not all users are as powerful.

A really good and concise cheat sheet for XDG is available on [Github as a Gist](https://gist.github.com/roalcantara/107ba66dfa3b9d023ac9329e639bc58c) 
I recommend adding this link to the Paths page.

jordanbrown0, 2025-09-15:
>>I'm not really familiar with MacOS, but from context I suspect that there is no one answer,

>actually there is one answer .. the /Applications directory is the default installation folder. It can be changed, but regular users would not bother.

This point is with regard to the user's documents directory.  It seems (again, based on what's been said here and a few Google searches) that Apple has decided that the right way to find that is by querying this NSDocumentDirectory parameter.  Perhaps there's a more human-friendly way to talk about that, but assuming that it has a constant pattern appears to violate Apple's intent.  It would not be shocking, for instance, if at some point in the future a user who speaks Norwegian were to get a directory named Dokumenter instead of Documents.

Also, it would not be at all surprising if the value of that parameter does not track $HOME, that $HOME is initially set to the parent directory, but that setting $HOME doesn't change application behavior.

>> Windows is definitely like that: the path to the user's documents directory is controlled by the system

> well .. it is setup by the system for the user but can be overridden using the Environment Variables app

Nope.  There is no Windows environment variable that points at the Documents directory, and changing environment variables does not change (correct) application behavior.  You might think that `%USERPROFILE%\Documents` would be the answer, but you would be incorrect.  I have a totally standard Windows install where I have opted to have my Documents directory be on OneDrive; my `%USERPROFILE%` is `C:\Users\Jordan` and my documents directory is `C:\Users\Jordan\OneDrive\Documents`.  And that's just using the simple OneDrive opt-in checkbox; you can change the Documents location arbitrarily using the "Location" tab on the properties for the "Documents" directory in Quick Access.

>> the current name is FOLDERID_Documents.

> not seeing where you get that name "FOLDERID" .. in my terminal environment i have

It's not an environment variable.  It's a system-supplied parameter, available from the Windows API.
https://learn.microsoft.com/en-us/windows/win32/shell/knownfolderid
Windows does **not** make heavy use of environment variables.

>> As for messing with the Registry... The danger is overblown. ... it's better to use the UI to manage them,

> my point exactly .. so i think the registry should not even be mentioned in our docs

The situation with the Registry is identical to the situation with configuration files on UNIX-y systems.  For both, it's better to use the UI to manage them.  And for both, there are occasionally good reasons to manage them directly.

In fact, just yesterday somebody asked how to dump their entire OpenSCAD configuration as a text file, and the answer was a Registry query.  Every once in a while somebody gets their configuration bollixed up - typically somehow ending up with really bad saved window locations - and needs to just blow away the relevant registry keys.

But also:  this page is mostly a reference for power users and developers.  Ordinary users will not mess with anything here.  Ordinary users, to the small extent that they care (e.g. to install libraries) should be looking at Help/Library info.  I've just added a note at the top pointing at that.