openscad/openscad #6210: Crash when trying do anything

cacheflood, 2025-09-16:
**Describe the bug**
Opening openscad with file on fedora 42 leads on crash.

**Environment and Version info (please complete the following information):**
6.16.7-200.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Sep 11 17:46:54 UTC 2025 x86_64 GNU/Linux

**Library & Graphics card information**
Nvidia 580.82.07

**Additional context**
Crash on odd places. Program stays alive for few seconds and then crash. when I open some example project, I can turn it few seconds and then program crashes.

(openscad:39556): Gtk-WARNING **: 00:16:31.161: Theme parsing error: <data>:1:0: Expected a valid selector
GLEW Error: Unknown error
The Wayland connection experienced a fatal error: Protocol error

airtonix, 2025-09-27:
same here.

Will launch.

If I do nothing and touch nothing, it dies after 10-20sec

If i mouse over a field that can take focus or hover styling, then it dies.

https://github.com/user-attachments/assets/4a1ee27b-cc14-4cda-8659-75d67bbad20d

kintel, 2025-09-27:
> `OpenSCAD version 2021.01` 

This is getting a bit long in the tooth by now. Could you try a recent snapshot and see if it's been resolved? https://openscad.org/downloads.html#snapshots

airtonix, 2025-09-30:
> > `OpenSCAD version 2021.01`
> 
> This is getting a bit long in the tooth by now. Could you try a recent snapshot and see if it's been resolved? https://openscad.org/downloads.html#snapshots

Sure, but for reference I also tried the flatpak and same thing happened.

airtonix, 2025-09-30:
ok the appimage works