openscad/openscad #6216: Can you make the F5 compile interuptible in case of very slow renders or warnings

Sasawaws, 2025-09-20:
Got a bit of code that works fine, slight typo and now it's spewing warnings - been doing it for several minutes, can't interrupt it. It's a bit annoying.

Same with very slow F5 renders

closed 2025-09-20 by jordanbrown0

MichaelAtOz, 2025-09-21:
This used to be a two step process, with one post manually explaining the duplicate, then a 'close'.
The result was two emails including the details of the duplication.
Now it is one email: "Closed [#6216](https://github.com/openscad/openscad/issues/6216) as duplicate."

[image]

Is the above close method some OpenSCAD Git automation or a new feature of GitHub?

t-paul, 2025-09-21:
I think it's a relatively new Github feature that the close buttons has a 3rd entry to close as duplicate with selection of the other issue

[image]

which marks it as such in the Github database too, not just having some sort of comment indicating the fact, e.g. in the issue list

[image]

jordanbrown0, 2025-09-21:
Yes.  I can be more verbose if people like, but "closed as duplicate of xxx" seems adequate.

MichaelAtOz, 2025-09-22:
Thing is in the email it says "Closed #6216 as duplicate." (ie the issue no. being closed).
The duplicate is missing thus requiring a visit to GitHub.
I was hoping it was automation that could be customised to include the same info shown in the post into the email.
That is the reason for the email, to email people the contents of the post. 
But in this case "Preview rendering in a separate thread" and its number is missing.
Does GitHub accept issues for itself?

t-paul, 2025-09-22:
> Does GitHub accept issues for itself?

I have not seen a way to create actual issues, but there's that: https://github.com/orgs/community/discussions/categories/projects-and-issues