openscad/openscad #6220: Ubuntu repositories

nomike, 2025-09-21:
github-ci-linux-get-dependencies.sh uses repositories providing "lib3mf" and "cgal" for Ubuntu focal (20.04) and Jammy (22.04):

https://github.com/openscad/openscad/blob/da8418b0b0a0219e29df4a4604034a919c515654/scripts/github-ci-linux-get-dependencies.sh#L15

Those repos lead to a HTTP 404 though and no longer seem to work.
I guess this will break builds on those old Ubuntu versions.

@t-paul You probably want to have a look at this.

t-paul, 2025-09-21:
No idea what happened on OBS. I think the 20.04 can be removed (#5666). For 22.04 we can try without those repos, according to the package list 22.04 has CGAL 5.4 and lib3mf 1.8.1 which should be ok if our README is correct.

nomike, 2025-09-21:
I will test building on Ubuntu 22.04 in a Docker container.