openscad/openscad #6226: non-manifold .obj output from manifold back-end

tturpin, 2025-09-24:
**Describe the bug**
The attached model, compiled with manifold back-end (without error), outputs a .obj file which causes a "NotManifold" error when re-imported in openscad

**To Reproduce**
`
openscad --backend manifold drill_bit_sharpener.scad -o drill_bit_sharpener.obj
`
Then compile (render) the following:
```
import("drill_bit_sharpener.obj");
cube(1);

```
either with CGAL or manifold back-end. With manifold, this fails with:
> Rendering Polygon Mesh using Manifold...
> WARNING: PolySet -> Manifold conversion failed: NotManifold
> Trying to repair and reconstruct mesh..
> ERROR: [manifold] Input mesh is not closed!
> ERROR: [manifold] Surface_mesh -> Manifold conversion failed: NotManifold

With CGAL, the error is:
> Rendering Polygon Mesh using CGAL...
> ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.

**Environment and Version info (please complete the following information):**
 - Ubuntu 24.04.2 LTS
 - System: Intel PC 64-bit
 - OpenSCAD master branch da8418b0b0a0219e29df4a4604034a919c515654

[drill_bit_sharpener.scad.txt](https://github.com/user-attachments/files/22516158/drill_bit_sharpener.scad.txt)

tturpin, 2025-09-24:
I will try, in the coming days

kintel, 2025-09-26:
Thanks, I can reproduce - will look into this one.